Tow-Boot/Tow-Boot #171: tag a new release with pinephone pro add smbios data patch?

ollieparanoid, 2022-08-16:
Hi,

it would be great to have a new release tagged that includes https://github.com/Tow-Boot/Tow-Boot/pull/135 since without that patch alsa can't find the proper UCM config path ([pmaports#1601](https://gitlab.com/postmarketOS/pmaports/-/issues/1601)):

```
pine64-pinephonepro:~$ cat /proc/asound/cards 
 0 [PinePhonePro   ]: simple-card - PinePhonePro
                      Unknown-UnknownProduct-
```

I'm not sure what the plans for the next release are, maybe it's worth backporting this on top of the previous release?

Thanks for all the work on Tow-Boot!

samueldr, 2022-08-17:
I'm preparing one by doing the trivial tasks I had in the TODO list, and will cut a release probably today, or else tomorrow.

samueldr, 2022-08-18:
I have still to get a few representative devices out to test the release, which I'll do tomorrow.

In the meantime, since you have a particular issue you want to test against, you can use this build built from the current tip of `development`:

 - https://stuff.samueldr.com/f7bspy0c3zk0c1x4mjiwj43i7rv7rjw5-Tow-Boot.release.2021.10-005-pre/pine64-pinephonePro-2021.10-005-pre.tar.xz

It should basically be the build used in the release, except with the `2021.10-005-pre` version build number instead of `2021.10-005`.

samueldr, 2022-08-19:
The new release is now up

 - https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2021.10-005

It should solve the underlying problem here.

closed 2022-08-19 by samueldr