Tow-Boot/Tow-Boot #174: Investigate enabling pstore by default

samueldr, 2022-08-18:
Ref:

 - https://u-boot.readthedocs.io/en/latest/usage/cmd/pstore.html

AndreySV, 2022-12-19:
BTW, have you managed to get working pstore on OG PP? ramoops?

AndreySV, 2023-07-29:
JFYI u-boot doesn't support compressed pstore yet. So any pstore compression algorithms should be disabled in the kernel config. By default DEFLATE compression is used. It looks like compression is used only for dumps. console and user logs are not compressed.

[Here ](https://salsa.debian.org/Mobian-team/devices/kernels/sunxi64-linux/-/issues/72#note_413122) are details, that needed to get pstore working on OG PP.

samueldr, 2023-07-30:
Haven't looked at the problem space at all yet, so no. It's more of a note of a thing to configure and make sure is working before stating it is working :)

AndreySV, 2023-12-30:
This is one of the changes needed for pstore: https://github.com/Tow-Boot/Tow-Boot/pull/291