Tow-Boot/Tow-Boot #175: Support for radxa zero ?

neofeo, 2022-08-19:
Hi, I am testing EFI with the current radxa zero and latest uboot and I wonder if the normal zero could be added since it's  going to be on the market for several years like zero 2 and it's  a decently capable board for light desktop. In fact, I got better like with it than zero 2 on EFI. 

I will test tow boot on zero 2 soon tho and probably  make a video is it does work correctly

samueldr, 2022-08-19:
Hi!

I don't have the hardware, so I cannot exactly add it "blind". Though I don't have the hardware for many boards that were added, so as long as a contributor can properly test (and ideally make the changes), it is highly likely to be possible.

Given the fact it's an Amlogic S905Y2, I expect it should mostly *just work*.

* * *

As for helping you do this? It should be possible. Though right this moment I can't.

This really shows that a "How to port when I don't know Nix" guide would really be helpful (#9).

If you're trying on your end, a *really incomplete* guide is available in the docs

 - https://tow-boot.org/development-guide.html

For now it's more of a few hints about how to get some knowledge about Nix, nothing about Tow-Boot though.

neofeo, 2022-08-19:
Many thanks @samueldr . I will try it to port it, let say, but for now, I will test the zero 2. Leave it open since I guess some others will like to know about tow boot on the zero once tow boot gets more popular

neofeo, 2022-08-21:
Okay, so far I got it working on the zero 2 with fedora arm64 EFI and like on s905y2 (zero) audio and the display driver aren't working correctly. I am talking with the baylibre guy in order to get it fixed down the road. So far it seems amazing.
I noted that you are not going  to support to use micro sd as spi instead of the emmc, that's  okay anyway. Once I get the display driver working at least I will make a video on this. Awesome guys.

samueldr, 2022-08-21:
> Once I get the display driver working at least I will make a video on this

Noting that with the radxa-zero2 on LTS 5.10 (IIRC) the display worked with the innate FDT (dtb built into Tow-Boot).