Tow-Boot/Tow-Boot #176: 2021.10-005: PinebookPro Error Flash firmware to spi

stacyharper, 2022-08-19:
I'm trying to install the last version of tow-boot to my PinebookPro. The bootloader was vanilla before that.

The installer fail :

```
---------
ERROR
---------

This is the installer for : [pine64-pinebookPro]
The board detected is: []

Unknown command 'paue' - try 'help'
```

Do I have a special board ?

Nimayer, 2022-08-19:
Hi @stacyharper, I encountered the issue you are reporting.

My situation was the following:
- eMMC with old postmarketOS installation
- old U-boot (not Tow-Boot) installed in SPI flash
- microSD with SPI installer flashed

I was able to boot the microSD by stopping the old U-boot startup with CTRL+C, but the Tow-Boot installer was not showing the Tow-Boot logo and reported the same ERROR as yours.

I fixed this by wiping the installation on the eMMC (zeroing the first 128MB) and rebooting the machine.
Strangely the Pinebook Pro would not turn on anymore after that so I had to press the reset button on the main board.

After pressing the reset button the Pinebook Pro booted the Tow-Boot installer from microSD, showing the Tow-Boot logo and I was able to install Tow-Boot.

I suggest disabling the eMMC with the hardware switch instead of wiping the eMMC to avoing hanging the Pinebook Pro as I did. 

I hope this helps.

samueldr, 2022-08-19:
@Nimayer's observations should be correct.

What we're observing here is that the U-Boot that launched the installer script is not "aware" of a Tow-Boot specific environment variable that is used by the installer script. This is also why the `pause` (shown as `paue` probably due to a letter being dropped by the serial interface) command cannot be found.

Note that even if the environment variable was present, the installer script uses features that are not turned on in U-Boot by default.

All of this is similar to #122, but not exactly the same. The same solution would fix both issues.

What needs to happen to *fix* this is to have a more robust installer experience. The hush-based script can't deal with unexpected scenarios gracefully.

samueldr, 2022-08-19:
Some additional notes for @Nimayer, probably irrelevant for @stacyharper at this point.

> Strangely the Pinebook Pro would not turn on anymore after that so I had to press the reset button on the main board.

It's possible it was "stuck" in what is commonly called "maskrom mode". In that state, the Pinebook Pro has no LED turned on, and is waiting patiently for a loader program to be uploaded via the USB port.

To quit "maskrom mode" without resorting to the reset button, one can hold the power button *longer than expected*.

To diagnose whether the system is in maskrom mode, connecting its type-c port to a computer should show something like this under `lsusb`

```
...
Bus 002 Device 062: ID 2207:301a
...
```

The VID being 2207 is the important part, the product identifier may differ.

sevens, 2022-08-19:
I had this issue as well, trying to install 2021.10-005:
* New Pinebook Pro, delivered last week
* Using a manually installed Manjaro-arm (which overwrote the stock install)
* U-boot was stock but one of the recent OS updates (or the manual reinstall?) updated u-boot as well; the update made the screen work during u-boot so it was quite noticeable
* Booting Tow-Boot installer from SD (via u-boot), a lot of text kept scrolling by (way too fast to read).  Could stop this via `Ctrl+c`; it was complaining endlessly about no menu system being available and some missing commands, telling me I could run the commands manually.
* I ran `run spi_flash` and after a second or 2 the error from this report showed up

@Nimayer's work-around (disabling eMMC, flashing SPI from SD and then enabling eMMC again) installed Tow-Boot properly, system now boots fine with it.

Note: in my case it actually said "pause" in the message (not "paue"), on the Pinebook's screen, no serial console used.

Nimayer, 2022-08-19:
> It's possible it was "stuck" in what is commonly called "maskrom mode". In that state, the Pinebook Pro has no LED turned on, and is waiting patiently for a loader program to be uploaded via the USB port.

Thank you @samueldr for the information about maskrom mode, that’s useful to know!

myself600, 2022-09-13:
I'm encountering the same issue. Disabling eMMC by flipping the switch near the chip did not help (same error).

petaramesh, 2022-09-24:
Hello, I fell on the same issue and found a workaround : 

My PineBook Pro has Manjaro installed with manjaro's u-boot on eMMC.

Is this setup, the “Shared storage installer” SD card of Tow-boot will simply not boot at all.

The “SPI installer image” would boot, have error messages scroll very fast until [Ctrl]-C, then I could start the installer script that failed with the exact message Stacy Harper reports : The board was not recognized.

After some hesitations I just erased Manjaro's u-boot from the eMMC with : 

dd if=/dev/zero of=/dev/mmcblk2 seek=64 count=334 conv=notrunc,fsync
dd if=/dev/zero of=/dev/mmcblk2 seek=16384 count=1956 conv=notrunc,fsync

Afterwards I turned OFF the Pinebook Pro, then powered it on the Tow-boot SPI installer SD-card, it booted as expected, chose to write to-boot to SPI that went fine and quick, rebooted et voilà !

The machine boots just fine, Tow-boot works just fine and my Manjaro is happy, nothing was lost.

Maybe the “just erase the eMMC u-boot” part with the relevant dd commands should be added to the PineBoo Pro page at : https://tow-boot.org/devices/pine64-pinebookPro.html ?

Thanks for the nice tool !

myself600, 2022-11-10:
Or `dd if=Tow-boot.spi.bin of=/dev/mtd0`

River-b, 2022-11-24:
> After some hesitations I just erased Manjaro's u-boot from the eMMC with :

hello, I'm having a horrible time 2 days now trying to get SPI with tow boot. not working I tried your method deleting the  manjaro boot  but it did not work.  I actually didn't even see the error messages you got.  when I flash  the  towboot Directly to the SPI  while Inside of  my  SD Armbian .   and then I reboot, I dont get any tow boot text at all.  I have never seen the tow boot Icon or text in 2 days of effort. 
do you have any idea what I could do?  
Also when Manjaro was working I tried to write  to SPI directly but on power off and on it still did not show a menu. 
Now with manjaro  uboot deleted  ,  I can not boot into the emmc  manjaro. 
When I try to boot with  Spi Installer  SD card,  it  has red power light,  and screen is =dark blue  (not black)  but no text ever appears and no tow-boot icon.  

I'm in  the Pine64  Pinebook  live chat room (Discord).   if you have time to drop by and helps us. thanks

petaramesh, 2022-11-24:
Try per : https://github.com/Tow-Boot/Tow-Boot/issues/196#issuecomment-1326477537

Also check that your Tow-boot SPI installer SD card is actually good...

River-b, 2022-11-24:
> 

> Also check that your Tow-boot SPI installer SD card is actually good...

1. **How do I check the SD card is good?**   after I make the SD card from inside the SD card with ARmbian on it, I see  the spi.imstaller file inside the  SD card.   I beleive  that the SD  cards are good because  I loaded also Armbian on all of them to check and Armbian boots up from SD card. but not SPI.installer  SD.
2. Note  that we also tried to install it  directly to SPI from inside  Manjaro 2 days ago and that did not work either.  No Tow-boot icon , no display of text or error messages. 
3. I made 5 different SD cards and even ordered  3 more new ones from  Amazon to try and  nothing  boots up to show on the screen.
4. I think Tow boot is there  perhaps, but I can't see it on display.   we used spi.Installer.img at this point and then  we also tried  writing to them  2 older versions to  Tow boot. 

5. to installed directly onto /dev/mtd0    we used the Tow-boot.spi.bin that was inside /binaries  folder  and also tried different  versions since the newest version did not work  we tried  10.005,  10.004,  and an 04-002 versions of Tow-Boot.  
6. when using the  SD card versions,  they don't boot up,  I never see any display of Tow boot icon or messages or menu.
7. When using the version written directly to  the SPI flash from within manjaro when it was work, we also tried different versions of it (Tow-Boot.spi.bin but different versions).

River-b, 2022-11-26:
TL_Lim  of Pine64  thinks that not seeing tow-boot on the Pinebook Pro Screen might be a bug  in Display driver of Tow-Boot.
This affects  some  versions of Pinebook Pro people.  
So far at least  2 of   us that have same exact problem confirmed we have   older Pinebook Pro  which has  a manufacturer marking date of  **2011**  on the Small_Board ( this is the PCB where the SD care is).   2011  means made in the 11th week of 2020.

`[T] <tl_lim> @River-mochi, from the chat log, I suspect is tow-boot's display driver bug. From your description, your PBP indeed boot up from the tow-boot SD card and your observe the green status LED. My takes the LCD display driver that tow-boot using may not have correct LCD timing and cause some LCD display cannot lock the display team and this is why no display. I think someone already file such issue on tow-boot GitHub and hopefully this issue can be resolved soon by update LCD driver. I think your PBP SPI Flash is OK and no need to desolder.
`

petaramesh, 2022-11-26:
Well if Tow-Boot actually boots, the power LED should show solid RED, not green...

A green LED makes me think that some other u-boot took over.

I would be surprised if the manufacturing date would make a difference. There have been minor changes on the PCB with time, but the SOC is still the exact same RK3399, and as there is no onboard firmware, well, either Tow-Boot can boot a RK3399, or it can't...

However I wonder if there might have been changes in the LCD panel, that's possible.

Anyway the green LED seems to indicate that Tow-boot is not running, if it were, the LED would be RED.

River-b, 2022-11-27:
> Well if Tow-Boot actually boots, the power LED should show solid RED, not green...

I had red light with tow boot but screen stayed  dark blue color and blank.  I think he is also referring to other tests that we all did in discord also.  we did many many tests and different things over 3 solid days.  including the suggestion you had of deleting  uboot with manjaro.
we also tried installing PostmarketOS  on the emmc and using that uboot and it also did not work. 
however another  person in  discord chat  had a special  Pine64 UART cable and said that  he  could  see on another screen using the UART cable ,  the PMOS  uboot menu.   this only further suggests   a problem with display driver used for tow-boot and also for PMOS uboot.
damnit  in discord said he could only get menu on 2nd screen  with UART cable if  he used PMOS uboot.   with tow-boot  he got error messages.   which is better than the empty blank dark blue screen  he sees on pinebook pro. 

it's a side thing that might not be related but it appears that both me and damnit  have  PBPRo made around March 2020.

myself600, 2022-11-27:
What about returning the device to Pine64 for investigation? They should cover these corner cases.

petaramesh, 2022-11-27:
I doubt it. I've been extremely disappointed by replies I had from Pine64 customer support for battery charging issues I had with my PB Pro - that ended up in a dead charger - with my PBP just about a month old.
So for a 2020 machine, you'd better not expect anything.

River-b, 2022-11-27:
not sure this is a pinebook pro problem.  it might just be a display driver issue that can be corrected on the software side.

River-b, 2022-12-09:
I see what you mean.  At this point a bunch of us  still can not see tow-boot if we try to install it on SPI.   Same with U-boot.  

One person with a Pine UART  cable that goes to the 3.5mm hole  reports that they can Only see something on 2nd screen  using the UART cable and can successfully  see   PMOS  uboot menu but ONLY  using the special cable to other monitor.