Tow-Boot/Tow-Boot #180: RP64 - unable to install/ no usb keyboard support

psstoyanov, 2022-08-21:
I'm unable to install [2021.10-005](https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2021.10-005) on RockPro64 as there is no support for USB keyboard devices.

Connected to:
- https://github.com/Tow-Boot/Tow-Boot/issues/153
- https://github.com/Tow-Boot/Tow-Boot/issues/147

I can't recall which version of uboot included keyboard support but I believe this is the most likely cause.

Note:
The operation can't be executed through UART - I can't find another ticket which outlines the need of such support (outputting to serial and accepting input).

Note 2:
 If using a newer uboot version, towboot will pipe to serial but the operation can't be completed:
 https://github.com/Tow-Boot/Tow-Boot/issues/176

samueldr, 2022-08-22:
> Having good instructions on how to safely wipe the SPI from within towboot might be warranted.

You might be interested to know that the installer has an option to erase the SPI flash.

psstoyanov, 2022-08-22:
Yes, the installer itself does and I agree that it is very easy to use.

The edge case which the suggestion attempts to address, is for the extreme cases where either the user doesn't have sd card available at the moment to use the installer or if for whatever reason there is incompatibility with future version (shouldn't happen given the extensive testing but it's a possibility).

Nuking the SPI with `sf erase 0 80000` from the `Firmware Console` menu option does the job but it doesn't feel like the "proper" solution (I will check if the code as of how the installer does it tomorrow).

psstoyanov, 2022-08-22:
Oh, I just saw where [Getting-started](https://github.com/Tow-Boot/Tow-Boot/blob/released/doc/getting-started.md) lives. Will see to create a PR for the baud rate tomorrow.

samueldr, 2022-08-22:
> Yes, the installer itself does and I agree that it is very easy to use.
> 
> The edge case which the suggestion attempts to address, is for the extreme cases where either the user doesn't have sd card available at the moment to use the installer or if for whatever reason there is incompatibility with future version (shouldn't happen given the extensive testing but it's a possibility).
> 
> Nuking the SPI with `sf erase 0 80000` from the `Firmware Console` menu option does the job but it doesn't feel like the "proper" solution (I will check if the code as of how the installer does it tomorrow).
> 
> Documented behaviour and procedure in extreme cases is safer than undocumented one even if it's ultimately never expected slightly_smiling_face

`sf erase 0 80000` would be a *sufficient* way to do it.

This is somewhat related to #47, though here it's about neutering on the dedicated storage instead of the shared storage.

psstoyanov, 2022-08-23:
Thanks, @samueldr ! This is _exactly_ what I was looking for: https://github.com/Tow-Boot/Tow-Boot/blob/00f2cce0e270f87e8f013e476ef5db8e0f79e3b7/support/builders/spi-installer/default.nix#L79 and it is indded closely connected to #47 

I'm not sure what would fit there - separate `Troubleshooting` page? :thinking: Anyway, neutering the SPI indeed is a better fit for the other ticket.

For the baud rate, I'm thinking a separate section in [Getting-started](https://github.com/Tow-Boot/Tow-Boot/blob/released/doc/getting-started.md) but will see how it will go once I start writing it. Most likely will post in a few pine64 channels for feedback on the draft before submitting a PR.

samueldr, 2022-08-23:
Hmm, the documentation needs a page tying things together.

The baud rate is already documented here:

 - https://tow-boot.org/serial.html

psstoyanov, 2022-08-23:
This does give me a few ideas :100: 

I do feel like there should be at least a small paragraph in Getting Started stating the baud. The paragraph can have the links for further reading.

psstoyanov, 2022-08-23:
This is my draft: https://github.com/psstoyanov/Tow-Boot/commit/ec0e47869a7dcd8571078081ef972c86ee22326e

I'm waiting for a bit of feedback in the Pine64 channels before creating the PR.

@samueldr , if there is something you would like changed before I create it, please, tell me :+1:

jirutka, 2022-10-03:
Can you please try the patch from #199 (add it to pine64-rockpro64)? It should fix the problem with USB keyboards even on RockPro64.

psstoyanov, 2022-10-03:
Hi, @jirutka , I should've updated this ticket previously.

Have you seen my notes on #178 ?
Those two comments in particular:
- https://github.com/Tow-Boot/Tow-Boot/pull/178#issuecomment-1235858198 
and:
- https://github.com/Tow-Boot/Tow-Boot/pull/178#issuecomment-1251040661 

What's the difference with #199 vs the current commit on #178 ?

jirutka, 2022-10-03:
Yes, I read your comments.

As I wrote in https://github.com/Tow-Boot/Tow-Boot/pull/199#issuecomment-1264827408, I’ve tested #178 with no success, USB keyboard still doesn’t work.

I don’t have RP64, so I cannot verify how it behaves there. However, the fix I’ve used for ROCK Pi 4 is originally for RP64 ([source](https://www.mail-archive.com/u-boot@lists.denx.de/msg371891.html)). It’s interesting that `CONFIG_USE_PREBOOT=y` is already enabled in [rockpro64-rk3399_defconfig](https://github.com/u-boot/u-boot/blob/v2021.07/configs/rockpro64-rk3399_defconfig#L15) (unlike [rock-pi-4-rk3399_defconfig](https://github.com/u-boot/u-boot/blob/v2021.07/configs/rock-pi-4-rk3399_defconfig)), but `CONFIG_PREBOOT="usb start"` is not there.

psstoyanov, 2022-10-03:
The results from my RP64 will depend on which version it was. iirc earlier units displayed instability with `PREBOOT` . That's also mentioned in the Gentoo [wiki](https://wiki.gentoo.org/wiki/PINE64_ROCKPro64).
I can't be sure if my unit experienced the issue but I have some hazy memories that it did (don't hold me on it tho) or if the problematic behaviour can even be experienced with the newer versions of uboot.

Will give it a shot and add a reply here if there is a regression compared to #178

piegamesde, 2023-06-30:
As far as I can see, this issue is fixed on master and can be closed. Also, a new release would be nice :)

samueldr, 2023-07-06:
Assuming this is fixed, if *this* issue as experience by individuals here is still present, please reopen/ask to reopen. Otherwise, if experiencing something vaguely similar, or really similar, don't hesitate to open a new issue with details!

closed 2023-07-06 by samueldr