Tow-Boot/Tow-Boot #183: UEFI boot on roc-rk3399-pc doesn't work with any distribution

shvetsnikita, 2022-08-27:
Hello!

I was able to successfully flash 2021.10-005 release to SPI of roc-rk3399-pc, but I was not able to boot any UEFI iso (although booting Armbian images works). I getting into GRUB2 of Debian 11.2/12, Fedora 36/37 and openSUSE Leap 15.4 images, however I can not get to the installer itself. Debian and Fedora always stuck right after GRUB2 (black screen or "Booting a command list" if I trying to add nomodeset to kernel boot options) while openSUSE clearly attempt to boot kernel (screen flicker, so I guess rockchipdrm driver is trying to configure display; I also seeing same behavior with Manjaro ARM images) however in all three cases I get black screen and power down after some time.

So is this some Tow-Boot/uboot bug or just poor/regressed board support by upstream and/or distributions?

Strit, 2022-08-27:
Same happens for me on Pinebook Pro. It loads a functioning grub, but after that it just stops loading anything.

I tested with the Fedora Workstation 36 raw image.

psstoyanov, 2022-09-16:
Still a bit early as I didn't check with TowBoot specifically however I am booting with systemd-boot successfully on my Pinebook Pro.

Most likely I'm not getting the device tree stuff as the display turns off right when the kernel loads however it does boot successfully with UEFI :slightly_smiling_face: 
The TowBoot check will be done tomorrow.

psstoyanov, 2022-09-17:
Great success!

Booting in UEFI mode with TowBoot + Systemd Boot.
After the initial setup, install the kernel with:
```
kernel-install add <version> <kernel_image> <initramfs>
```

For Manjaro ARM this looks like so:
```
kernel-install add 5.19.8-1-MANJARO-ARM /boot/Image /boot/initramfs-linux.img
```

No idea why I had to pass exactly the version that is `/usr/lib/modules` but yeah.
If you **aren't** using TowBoot, most likely you would have to pass the devicetree in the loader entry like so:
```
title      Manjaro ARM
version    5.19.8-1-MANJARO-ARM
machine-id 2cc62178df8042989699bd78ff0f842f
options    console=ttyS2,1500000 root=PARTUUID=2724a9ad-2c3a-4fe1-a3bb-b437422731d3 rw rootwait audit=0
linux      /2cc62178df8042989699bd78ff0f842f/5.19.8-1-MANJARO-ARM/linux
devicetree /dtbs/rk3399-pinebook-pro.dtb
initrd     /2cc62178df8042989699bd78ff0f842f/5.19.8-1-MANJARO-ARM/initrd
```
(I've copied over the dtbs to my `/efi` dir for my first attempts)

**With** `TowBoot` however, you don't need to pass or have the dtb in the `/efi` dir - it's passed as expected by `TowBoot`. Example entry in `/efi/loader/entries`:
```
title      Manjaro ARM
version    5.19.8-1-MANJARO-ARM
machine-id 2cc62178df8042989699bd78ff0f842f
options    console=ttyS2,115200 root=PARTUUID=2724a9ad-2c3a-4fe1-a3bb-b437422731d3 rw rootwait audit=0
linux      /2cc62178df8042989699bd78ff0f842f/5.19.8-1-MANJARO-ARM/linux
initrd     /2cc62178df8042989699bd78ff0f842f/5.19.8-1-MANJARO-ARM/initrd
```

Notice that I've changed the ttyS2 baud so that it will match TowBoot :slightly_smiling_face:

shvetsnikita, 2022-09-18:
@psstoyanov did you tried to boot images with GRUB2? 

I using Tow-Boot of course (which is why I filled this bug report) and I tried to pass several different device trees, which didn't lead my anywhere. I also tried a couple of other PSU, tried to boot without attached USB cable, etc. 

Just tried [vendor image](http://distro.libre.computer/ci/ubuntu/22.04/) which is also does not work (but only with Tow-Boot). However, this one managed to display boot log for several second before board shut down itself (however this image managed to somehow corrupt root partition during such shutdown, which didn't happened with other images).

psstoyanov, 2022-09-18:
I've had similar trouble with Fedora on my PBP. After the install is completed, it reaches Grub but once you try to boot an entry, you get no feedback.
It could be that I didn't wait long enough, I haven't followed the instructions in the [docs](https://tow-boot.org/distributions/fedora.html) or something else. It's been some time since I had to waste time dealing with grub and I would like to avoid it if possible :slightly_smiling_face: 

I haven't tried Debian since I learned Tow-Boot (so a few weeks ago).
No idea about Ubuntu - I've lost the thread there long ago.

While inconvenient, trying Arch or Manjaro with systemd-boot (or one of your preferred distros - Fedora/ Debian) can be a good check to eliminate the possibility of problems with your specific board.
Note:
With systemd-boot , the `kernel-install add` command likes to set the wrong PARTUUID if you are installing from chroot. Check the UUIDs in `/efi/loader/entries/` afterwards. I haven't experienced this issue from booted system.

I know @Strit used Grub successfully yesterday with Manjaro and TowBoot `005`.
Personally, after the whole experimentation phase as systemd-boot is new to me, I find systemd-boot to be far easier to deal with than Grub. Even if there is less information about it

shvetsnikita, 2022-09-19:
> I've had similar trouble with Fedora on my PBP. After the install is completed, it reaches Grub but once you try to boot an entry, you get no feedback.

On PBP you at least was able to complete install, while in case of roc-rk3399-pc:

> I getting into GRUB2 of Debian 11.2/12, Fedora 36/37 and openSUSE Leap 15.4 images, **however I can not get to the installer itself**

moreover:

>  in all three cases I get **black screen and power down after some time**

So this is certainly not similar case in my opinion. 

As for systemd-boot I guess it makes sense to try it. I will try to do it later after checking situation with aml-s905x-cc and [2022.07 PR](https://github.com/Tow-Boot/Tow-Boot/pull/178) as I don't have any board with have good Tow-Boot support on hands right now.

shvetsnikita, 2022-11-06:
A couple of more datapoints:

1. [Image by LibreTech](http://distro.libre.computer/ci/ubuntu/22.10/) (with heavy patched u-boot 2022.10rc5 and even more heavily patched Linux 5.19.17) boot in UEFI mode.
2. Image from [this page](https://sd-card-images.johang.se/boards/roc_rk3399_pc.html) fail in the same way as Tow-Boot right now, so I guess it's something broken in upstream (probably in both of u-boot and kernel).

shvetsnikita, 2022-11-06:
Just want to let you know that I was able to boot regular Linux 6.0 build from Debian repository with LibreTech's u-boot in UEFI mode. So it's either something that in their uboot patches or in DeviceTree they ship with their u-boot build. Interestingly, if I boot same kernel with it's own DeviceTree I get same power down I seeing with Tow-Boot right now.

samueldr, 2022-11-06:
Oh, they have changes for the device tree

 - https://github.com/libre-computer-project/libretech-u-boot/commits/u-boot/v2022.10/roc-rk3399-pc/arch

We probably should import the vendor DTs here, as they are being written for contributing back to U-Boot anyway. It's not a situation where vendor code is icky and old and outdated.

samueldr, 2022-11-07:
Note that with #210 I verified the NixOS UEFI iso was able to boot [to stage-1].

shvetsnikita, 2023-08-02:
Not reproducible anymore with Tow-Boot 2022.07 and 2023.07.

closed 2023-08-02 by shvetsnikita