Tow-Boot/Tow-Boot #189: Tow-Boot on Pinephone beta edition

Saddamus, 2022-09-04:
Did anyone managed to make tow-boot working on pinephone beta edition ?
I tried releases for pinephone and pinephone pro, flashing to SD card, but none of them was working. There is no reaction on pressing power button - not even blinking led light.

Saddamus, 2022-09-11:
This was my mistake - incorrectly flashed file.

closed 2022-09-11 by Saddamus

samueldr, 2022-09-11:
Can you say how it was incorrectly flashed?

Was there information that could have helped you that was missing from the documentation?

Saddamus, 2022-09-11:
I' am TL:DR kind of human. I have flashed several distros with Raspberry PI Imager, which takes XZ archives and manages them by itself. I didn't care what is inside of your archive - assuming it should work as with any other distos. Unpacking the archive and taking the right file to flash SD card resolved my issue :)