Tow-Boot/Tow-Boot #191: docs: pinephone: how to update the mmcblkboot install

colemickens, 2022-09-18:
I think the "installer" method of pinephone's payload installs to the mmcblk0boot* partitions?

Is there a quick guide on how to update it in place? Or do you recommend re-booting the installer?

samueldr, 2022-09-18:
For now it's using the installer.

`fwupd` is being worked on mainly by @DylanVanAssche, so with a lot of handwaving, it will eventually be a component updated by the system *just like your old boring BIOS on your old boring computer*.

DylanVanAssche, 2022-09-19:
fwupd support will come eventually, I already made some progress on it
and upstreamed some parts. However, you will need to use the installer
image until Tow-Boot contains version information as uSWID data.