Tow-Boot/Tow-Boot #196: [Pinebook Pro] Can only boot from SD after power cycle

petaramesh, 2022-09-27:
Hello,

This is a strange one : 

I can only tow-boot my Pinebook Pro from SD card after a complete power cycle.

On the other hand, if I just reboot it (whether or not it had been previously booted from SD card or other) and select [Boot from SD],
I will always get : 

``
switch to partition #0, OK
mmc1 is current device
** No partition table - mmc 1 **
Couldn't find partition mmc 1:1
SD Boot failed.
Press any key to continue...
``

OTOH, after a complete power cycle, I get : 

``
switch to partition #0, OK
mmc1 is current device
Scanning mmc 1:1...
253 bytes read in 5 ms (48.8 KiB/s)
1:            Manjaro ARM
Retrieving file: /initramfs-linux.img
16911252 bytes read in 1779ms (9.1 MiB/s)
Retrieving file : /Image
``

...and then it boots !

Any clue ?

eyduh, 2022-11-11:
+1

But for any boot option. Basically I need to reboot a few times no matter which way I want to bit before it finds the boot partition/extlinux conf

vincele, 2022-11-20:
Hello,

Looks like I cannot reproduce your problem.

I have tow-boot 2021.10-005 on an SD card in a pbp without emmc, and with empty SPI.

I rebooted a few times, and it always booted properly (Arch Linux arm).

I tried:
- no kbd interaction, let the timeouts choose the default (multiple tries, always working)
- choose one boot option
- CTRL-C, then choose SD card, then one boot option
- ESC, then choose SD card, then one boot option

petaramesh, 2022-11-21:
Hello,

I have Tow-boot installed on the SPI, on a PBpro with both an eMMC and an NVMe, so there can be a difference compared to Tow-boot installed on an SD card.

If it is installed on an SD-card and loads, obviously the ROM firmware must have properly initialized the SD card subsystem for it to load.

OTOH, if Tow-boot starts from SPI, there seems to be some difference.

The fact that, after a full powercycle, I can always boot from SD, but after a reboot, I get the errors reported above, is completely systematic and reproductible here (regardless of what device I last booted from).

vincele, 2022-11-21:
Hello,

Yeah I only meant it as "just another data point". It may still help to know that it works properly in this setup.

River-b, 2022-11-24:
> 

> I have since flashed the u-boot from postmarketOS to the SPI and can now choose which boot option I want without power cycling.
@eyduh 

hello I am new to PBPro,  and having a lot of issues trying to  install tow-boot, from the SD card, the menu never appears.  I have manjaro  on the Emmc  and it can boot into that.  I noticed you said you got a different U-Boot from  postmarketOS? 
is it possible for me to use this save U-Boot?  I have been trying to get  tow-boot  to install directly on SPI  from inside manjaro  and also to boot  different SD with Tow Boot  and nothing works. 
may hours now no sleep.   
if  this  PostmarketOS u-boot works.  that would be excellent, but I have manjaro installed and don't know if there is any way I could install  Postmarket without  a tow boot working. 
Please let me know I would appreciate.

vincele, 2022-11-24:
The SDCard bootloader not working, is probably because you have a bootloader on emmc which takes precedence. The priority ordering is : SPI, eMMC, SD. So as long as there is something on one of the higher priority devices, you will not see what is on the lower prio ones.

eyduh, 2022-11-24:
> I have been trying to get tow-boot to install directly on SPI from inside manjaro
Not sure if this is possible, but [there is a section on the postmarketOS wiki](https://wiki.postmarketos.org/wiki/PINE64_Pinebook_Pro_(pine64-pinebookpro)) on how to do it from inside postmarketOS. Maybe the commands work in manjaro as well, I wouldn't know.

The way I installed tow-boot was to just open up the back of the pinebook pro and remove the eMMC, I then booted with the appropriate .img flashed to a microSD - after the flash procedure I put the eMMC back and closed up the back of the laptop.

(I needed to check out which microcontroller version I had for the keyboard & touch pad to know if I could flash it safely with update firmware and thus had to open up the back anyway so this was the easiest way to do it)

petaramesh, 2022-11-24:
All this is way outside of a bug report well...

But to install tow-boot on the SPI, you don't need to open the Pinebook Pro or change any switches.
You just need to first create the SPI install tow-boot SD card per instructions ;

Then erase the bootloader which is on the eMMC, from inside Manjaro, performing : 
- Clear the SPI : 
sudo flash_erase /dev/mtd0 0 0
- Clear the bootloader from the eMMC : 
dd if=/dev/zero of=/dev/mmcblk2 seek=64 count=334 conv=notrunc,fsync
dd if=/dev/zero of=/dev/mmcblk2 seek=16384 count=1956 conv=notrunc,fsync

Then you are sure you have no old bootloader remains on the Pinebook Pro.

Now you can just boot with the Tow-boot SPI installer SD-card inserted, and it will just work.

(Afterwards, your Manjaro will happily boot again)

River-b, 2022-11-25:
> The way I installed tow-boot was to just open up the back of the pinebook pro and remove the eMMC, I then booted with the appropriate .img flashed to a microSD - after the flash procedure I put the eMMC back and closed up the back of the laptop.

unfortunately those instructions dont work outside of PostmarketOS.   since I deleted the 2 lines of Uboot in manjaro it no longer boots the Manjanro on emmc.  but I can still boot to  SD card Armbian. 
I've made 5 different  SD cards now with   spi.installer  and none of them boot up when  I start the Pro.   when you have the emmc out.  do you  have switch  for emmc on or off.    
In 3 days now of  testing, I have  not gotted the  Tow-Boot Icon to appear a single time. I have not gotted any Text or menu or error messages. All I see is  screen goes from black to Dark blue when it powers on with the  SD ( spi.installer)  and then stays like that for a long time. 

Before I deleted the 2 lines of uboot in manjaro.   what happened is even though I tried to flash tow-boot SPI  to /dev/mtd0  , on power up,  it just hangs at Blank dark blue screen for a long time and then eventually  booted manjaro when it timed out I guess.   Now it doesn't do that since I deleted the uboot of manjaro. 
I can only boot with SD cards with  Armbian or Manjaro on them.  I can not boot with any SD cards made with SPI.installer (I mean I can boot, it's just a dark blue screen - blank for a long time and nothing happens).
Flashcp  directly to the  SPI  from inside  my SD-Armbian boot also did not work.

River-b, 2022-11-25:
> (Afterwards, your Manjaro will happily boot again)

once you got  Tow-boot working,  did you just simply reinstall OS on emmc  from an SD card Linux boot ?

petaramesh, 2022-11-25:
Once Tow-boot is running, it will happily boot the Manjaro the is already in the eMMC, you dont't need to reinstall anything else.

You could event convert in-place your Manjaro to UEFI bootingif you wish so, following these instructions : https://forum.manjaro.org/t/convert-existing-generic-install-to-generic-efi/124966/14

petaramesh, 2022-11-25:
Are you sure you  used the correct Twi-boot image for the Pinebook Pro ?
Because if you have an SD card containing the correct Tow-boot image, properly written, and you have removed all u-boot remains from your SPI and eMMC as I explained before, it should definitely boot the Tow-Boot SPI installer.
If it does not, I can imagine no other explanation that your Tow-boot SPI installer card is no good. 
Wrong installer file, or wrong installation method ?

The proper file for the Pinebook Pro is here : https://github.com/Tow-Boot/Tow-Boot/releases/download/release-2021.10-005/pine64-pinebookPro-2021.10-005.tar.xz

petaramesh, 2022-11-25:
The instructions are here : https://tow-boot.org/devices/pine64-pinebookPro.html

eyduh, 2022-11-25:
> unfortunately those instructions dont work outside of PostmarketOS. since I deleted the 2 lines of Uboot in manjaro it no longer boots the Manjanro on emmc. but I can still boot to SD card Armbian. [...]

Could you make a new issue cause this is unrelated to the issue OP and I was facing

repagh, 2022-12-19:
I have a similar issue, using the 2021.10 tow-boot on SPI:

- I can boot off the SD card, after selecting that with the menu
- When letting Tow-Boot start up and try the eMMC, that fails, and from my SD card OS (Debian) I cannot access the eMMC
- If I get to the menu in time, and prevent Tow-Boot from touching the eMMC, then after directly starting into the SD, I can mount and read the eMMC fine

Can this have anything to do with the mmc boot_mode?