Tow-Boot/Tow-Boot #197: Cannot boot with tow-boot on SPI on the RockPro64

callegar, 2022-09-28:
Hi,

I am trying to install debian using the debian installer for the RockPro64 on a RockPro64 with EMMC and the PCIe SATA Card.

Must say it is unclear to me if debian installed on EMMC using the installer needs or not an u-boot on spi or installs its own (can't find documentation in this sense), but I think so and debian does not boot with an erased SPI flash.

I have tried installing Tow-boot on the SPI. With it the system boots to the Tow-boot splash screen. However, trying to boot the OS results in a failure, notwithstanding the fact that tow-boot appears to find it on the emmc. Specifically the boot hangs at

```Booting using the fdt blob at 0x1f00000```

The same thing happens if I insert an SD card with a manjaro image in the SD card reader. Manjaro provides its own u-boot though and boots regularly without tow-boot on the SPI.

It is unclear to me if I am running into bug #186 that also mentions a freeze at booting using the fdt blob. Comments there suggest that the system should boot once the USB devices are disconnected but on my system disconnecting the USB devices seems to make no difference. Any clue? Please see also https://forum.pine64.org/showthread.php?tid=17406

shvetsnikita, 2022-10-07:
Please take a look: https://github.com/Tow-Boot/Tow-Boot/pull/178#issuecomment-1234631227

neofeo, 2022-11-01:
So, how to erase the spi without uart? Does tow boot provide anything  at all? Bc I face the same issue and for some reason, towboot doesnt recognize my usb keyboards, nothing, I can't  get into console and erase it. Please, help.

bendlas, 2023-01-30:
I had a similar issue, where my RockPro64 would hang at that message, unless I unplugged my USB keyboard.

I can confirm that upgrading to the develop branch of Tow-Boot (u-boot 2022.07) fixed the problem for me.

suirzes, 2023-02-14:
Same problem

jmevel, 2023-07-03:
> I had a similar issue, where my RockPro64 would hang at that message, unless I unplugged my USB keyboard.

Wow I can't believe this worked! I'd have never guessed that I think.

Any idea when the fix can be included in a new release?

jmevel, 2023-07-03:
> I had a similar issue, where my RockPro64 would hang at that message, unless I unplugged my USB keyboard.
>
> I can confirm that upgrading to the develop branch of Tow-Boot (u-boot 2022.07) fixed the problem for me.

Wow I can't believe unplugging my USB Keyboard during boot worked !
I'd have never guessed that I think.

Any idea when the fix can be included in a new release?

bendlas, 2023-08-06:
I think this should now be released as https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2022.07-006
Pending verification ...

jmevel, 2023-08-07:
> I think this should now be released as https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2022.07-006 Pending verification ...

Awesome, I'll try this new release when I have a moment.
Thanks for letting us know !

shvetsnikita, 2025-08-17:
This issue is not reproducible with 2023.07-007 release.

jmevel, 2025-09-01:
Indeed, I didn't think of coming back here but I also don't experience this issue anymore.
Whoever has the rights to close this ticket can do it I think.