Tow-Boot/Tow-Boot #198: How easy would pulling in a device from upstream uboot for tow-boot support be, if it already compiles and works?

CheetahPixie, 2022-10-02:
Because today, I had a thought.
https://github.com/u-boot/u-boot/blob/master/board/samsung/smdk5250/Kconfig#L14
If it is a relatively trivial-effort import, this might introduce Exynos support to some capacity.

samueldr, 2022-10-02:
The main difficulties come not in building U-Boot, but in producing the "user-friendlier" images. E.g. putting the built U-Boot at the right offset, when it's at an offset, and/or installing to the proper storage.

I don't know much about those older Exynos-based chromebooks, so I can't say for sure whether there are gotchas or something else. One thing that would help is some form of documentation detailing how one builds a mainline U-Boot and runs it on their system.

(I'm assuming here it's for the "snow" and "spring" chromebooks you're looking for, other Exynos systems might have different requirements.)

samueldr, 2022-10-02:
To answer further, let's look at Allwinner stuff.

If you look at the Pine A64-LTS, it's mostly metadata (to build the device page) and some light device-specific information. The most important one is the defconfig

 - https://github.com/Tow-Boot/Tow-Boot/blob/2f8fff7be70ebfd22098ee6a351b0075f63955fc/boards/pine64-pineA64LTS/default.nix#L15

When the Tow-Boot infra builds U-Boot here, it "just" builds U-Boot to get the actual build...

... but that's not where the story ends.

If you look under the `modules/hardware` directory, there is some further "knowledge" about the platform defined:

 - https://github.com/Tow-Boot/Tow-Boot/blob/2f8fff7be70ebfd22098ee6a351b0075f63955fc/modules/hardware/allwinner/default.nix

For example, for shared storage, this is where we define where the firmware is written to when building the shared disk image:

 - https://github.com/Tow-Boot/Tow-Boot/blob/2f8fff7be70ebfd22098ee6a351b0075f63955fc/modules/hardware/allwinner/default.nix#L71-L75

And this is how we get the right firmware file:

 - https://github.com/Tow-Boot/Tow-Boot/blob/2f8fff7be70ebfd22098ee6a351b0075f63955fc/modules/hardware/allwinner/default.nix#L76-L78

And to further close the loop, the platform configuration ends-up resolving the configuration for this:

 - https://github.com/Tow-Boot/Tow-Boot/blob/2f8fff7be70ebfd22098ee6a351b0075f63955fc/modules/tow-boot/disk-image.nix#L88

Which is then given to the disk image builder:

 - https://github.com/Tow-Boot/Tow-Boot/blob/2f8fff7be70ebfd22098ee6a351b0075f63955fc/modules/tow-boot/disk-image.nix#L59-L61

So really, the difficulties are not in getting U-Boot built, but in wrangling the build results into the right shape. In U-Boot this is generally dependent on the overall platform. Since it is dependent on the platform in U-Boot, it generally means that we need that knowledge written-up for every platform.