Tow-Boot/Tow-Boot #202: efibootmgr cannot write EFI variables on Pinebook Pro / Manjaro

petaramesh, 2022-10-24:
Hi,

I've noticed that efibootmgr fails fixing UEFI bootloader path and boot order on my Manjaro with tow-boot EFI.

It's annoying as it seems that tow-boot will only load the bootlader as /EFI/BOOT/BOOTAA64.EFI where Manjaro (and possible other Linux distros) will typically install it into /EFI/manjaro/grubaa64.efi

It's normally efibootmgr's job to define where to boot from into the UEFI variables, but here it fails.

So for the machine to boot, the grub binary has to be renamed so tow-boot can find it.

After some digging I found out that u-boot would be supposed to store the UEFI variables in a ubootefi.var file on the UEFI partition, but the way it actually stores them (or not) depends on some compilation flags I guess (config EFI_VARIABLE_FILE_STORE).

So I wondered how tow-boot is compiled and how is one supposed to access and update it UEFI parameters.

Thanks in advance.

samueldr, 2022-11-01:
Hi, sorry for the delay answering.

This is a known issue with U-Boot (and thus Tow-Boot).

The best upstream documentation bit detailing the issue is this:

 - https://u-boot.readthedocs.io/en/latest/develop/uefi/uefi.html#executing-the-boot-manager

So in other words, once the operating system is in control, U-Boot is unable to save EFI variables.

Note that AFAIK there is a technical reason for this, which I will simplify: having both the EFI services and the operating system "own" the device that holds the EFI vars at once is problematic, if not impossible. Let's think about EFI vars stored on the eMMC. The operating system is "driving" the eMMC device, and thus expects it to be in a coherent state, just like it was last time it looked at it.

This is a known issue, but an upstream issue.

petaramesh, 2022-11-26:
Hello,
I've tested and seen that the use of an EFI shell (outside of Tow-boot) will allow Tow-boot to store is EFI boot variabes onto the eMMC ESP.