Tow-Boot/Tow-Boot #203: Are Pinebook Pro DTBs jup to date in EFI mode ?

petaramesh, 2022-10-25:
Hello,

I just moved my Pinebook Pro's Manjaro from legacy syslinux-like boot to EFI boot.

I works well, but I've noticed that my system has become very notceably slower when booted if EFI mode compared to when booted in legacy mode - using the same kernel version.

I've also noticed that the devices (eMMC and SD card) are not numbered in the same order.

So I was wondering if the hardware description and settings provided by tow-boot EFI are the same as those provided by DTBs when using legacy boot ans if there are up-to-date ?

Seeing that tow-boot is apparently based upon an u-boot release that is already one year old.

Is there some update planned ?

Should I consider normal that the system would be slower when booted in UEFI mode compared to legacy boot ?

Thanks in advance.

Best regards.

Strit, 2022-10-28:
A quick search in the u-boot repository, reveals that the Pinebook Pro DTS found in the 2021.10 release of u-boot, which is what the latest release of tow-boot is based on, was last synced with mainline linux 5.14-rc1 on August 12th 2021.

See https://github.com/Tow-Boot/Tow-Boot/pull/178 for an planned update to 2022.07 in the works. This will bring in a DTS synced with 5.18 in June 2022.

samueldr, 2022-11-01:
I try syncing newer and "more exciting" platforms on top of the usual syncs by upstream U-Boot.

For the Pinebook Pro I had it synced to `20 Feb 2022`. Annoyingly I didn't note the Linux commit.

 - https://github.com/Tow-Boot/Tow-Boot/blob/2f8fff7be70ebfd22098ee6a351b0075f63955fc/boards/pine64-pinebookPro/0001-arch-arm-dts-Sync-rk3399-pinebook-pro-with-Linux.patch

The Pinebook Pro, though [is thankfully less "exciting" these days](https://github.com/torvalds/linux/commits/master/arch/arm64/boot/dts/rockchip/rk3399-pinebook-pro.dts). Or in other words, we're seeing basically no functionality changes. AFAIK there should be functionally no difference with the current state as of right now. The three tipmost current commits are mostly cleanup.

closed 2022-11-03 by samueldr

petaramesh, 2022-12-14:
Well I tried both and so far the “legacy” boot gives correct performance (I mean what you would expect given the machine's specs) and the EFI one really, really crawls.

But I would be very interested in your own independant impressions on the matter.