Tow-Boot/Tow-Boot #204: How to check Two-Boot starts?

Narann, 2022-10-25:
Hello,

First, thanks for putting you efforts in this project.

I read the doc here and follow every steps. At en end, there is:

> Once installed, remove the installation media, and verify Tow-Boot starts from power-on.

I remove the SD card (the "installation media"?) and rebooted. And it reboot normally, on the current OS, nothings seems to be different.

How can I check Two-Boot is actually installed?

Thanks in advance!

kirsle, 2022-10-25:
I could tell tow-boot is installed by seeing how many colors the LED light flashes during boot. The stock Pinephone bootloader only used one LED color (might've been red or something, I don't recall).

With tow-boot the LED on the Pinephone/Pro first turns red and then switches to a secondary color depending on what's booting: yellow for internal eMMC, cyan for microSD card (holding Volume Down on boot), blue for USB Mass Storage mode (Volume Up on boot). If your phone is doing all of this then tow-boot looks installed!

Narann, 2022-10-25:
Thanks, I realize the Red then Yellow/Cyan/Blue had something to do with Tow-Boot, but was not sure. Thanks for the support. :)

closed 2022-10-25 by Narann