Tow-Boot/Tow-Boot #209: 2022.07-development: USB problems on libreComputer-rocRk3399Pc

samueldr, 2022-11-05:
> I have added fixes that should now fix USB with the 2022.07 upgrade.

Just tested "feature/u-boot-2022.07" and "development" branches. On both branches USB Type-A ports works now (I didn't tested Type-C ports yet) although both throw USB-related errors in both cases, even if no USB devices is attached ("Failed to get clocks"). However, there is severe regression on development branch (I flashed it to SPI) - right after "Please press ... to enter the boot menu." Tow-Boot starting to scan mmc, print stack trace and then reboot. I can get into the menu if I press Esc soon enough, but even if I choice USB I still getting same outcome. Yet somehow spiinstaller works fine, I seeing issue only when I have Tow-Boot installed in SPI (shared image was not tested).

_Originally posted by @shvetsnikita in https://github.com/Tow-Boot/Tow-Boot/issues/178#issuecomment-1304636360_

samueldr, 2022-11-05:
A few things to check, which config are you using?

 - libreComputer-rocRk3399Pc
 - libreComputer-rocRk3399PcMezzanine

I should `diff` the configs, upstream hasn't been keeping them in sync.

Have you tried erasing SPI flash entirely? I'm asking since if you e.g. had saved the environment to SPI flash, *maybe* that could have saved values use within some drivers or elsewhere that are now wrong... Environment use in U-Boot isn't really great.

samueldr, 2022-11-05:
(Note to self, read https://github.com/Tow-Boot/Tow-Boot/issues/183 more closely and see if it is actually related)

shvetsnikita, 2022-11-05:
> which config are you using?

Without Mezzanine.

> Have you tried erasing SPI flash entirely?

Just tried to erase SPI entirely, then flash same build again - no any changes in behavior (still USB errors even when no USB devices is attached, even more error if USB dock is attached, crash and reboot after "Please press" message.

> Are the crashes happening only with USB devices connected? 

No, it crash without USB devices too.

samueldr, 2022-11-06:
Thanks for answering. These mainly help me determine what I need to test.

I am testing with the mezzanine config, since *in theory* they should have been the same, except for the dtb name built into the config. Right now, as-it-is, there seems to be no differences that should matter in the configs. The main difference in the config seems to be related to PCI, all of them.

The dts *seem* to only differ with features relevant to the mezzanine board.

I will need to dig up the device and test with the non-mezzanine build.

As for the SPI build not working right now, I wonder if it's related to the following:

 - https://github.com/Tow-Boot/Tow-Boot/blob/a5d2ea8ad9421ed70c736968535168fa712d982d/boards/libreComputer-rocRk3399Pc/default.nix#L32-L34

* * *

Something else I'm thinking about. Though it's more about SPI flash.

This board has a weird history. It was a project between two different vendors.

I got it from Libre Computer, from the crowd funding campaign. I assume this is the first "final" variant, if there are any variants.  I wonder if the other vendor (Firefly) made changes with theirs.

Did you buy yours from Libre Computer (loverpi counts as Libre Computer), or from Firefly (and resellers of Firefly)?

Can you read the exact model name of the SPI flash on the board? Or *if you can* take good quality pictures of its identifier marks?

shvetsnikita, 2022-11-06:
> Did you buy yours from Libre Computer (loverpi counts as Libre Computer), or from Firefly (and resellers of Firefly)?

From loverpi, exact model is ROC-RK3399-PC-V1.1 2018-09-25.

> Can you read the exact model name of the SPI flash on the board? 

I don't see where it's on the board. I checked [spec file for Plus model](https://download.t-firefly.com/%E4%BA%A7%E5%93%81%E8%A7%84%E6%A0%BC%E6%96%87%E6%A1%A3/%E5%BC%80%E6%BA%90%E4%B8%BB%E6%9D%BF/ROC-RK3399-PC%20Plus%20-%20Specification.pdf) (pdf for regular modem seems like is not available anymore or never was published on Firefly web-site) and [old product age](https://en.t-firefly.com/product/rocrk3399pc) where I at least spotted SPI letters on the bottoms of the board, but I still don't see where is SPI chip itself.

> Or if you can take good quality pictures of its identifier marks?

My phone camera probably is not good enough, but I could try to zoom-in relevant part of the top side of the board (if SPI is on the top of ofc).

> Also a good picture of the top and bottom of the board would help figuring things out about the provenance.

Board is assembled into the [case](https://www.loverpi.com/products/libre-computer-aluminum-case-for-pc-form-factor) with custom cooling solution, so if it's possible I would like to avoid reassembly. But if it will be really necessary (for instance if exactly same build would work on V1.0 but crash on V1.1) I will do that.

samueldr, 2022-11-06:
Right, no worries, it was if you could. If you bought from Libre Computer, then it's the expected one.

The "Pro" and "Plus" from Firefly should be seen as different products for what matters here.

So I guess I'll have to try the non-mezzanine build, to see if it works on mine. If it still does, I guess the other thing to try would be disabling the SPI speed changes. Other than that, I wouldn't exactly know what to look for. E.g. I would have to first look at stock U-Boot (same version), then previous versions, and then once a first bad version is found, bisect to hopefully find a culprit.

shvetsnikita, 2022-11-06:
Just in case, build from your "feature/u-boot-2022.07" tree still works from SPI without crashes. So it's something that was added in development branch very recently.

samueldr, 2022-11-06:
@shvetsnikita do you have the exact commit? I force-pushed a few times in-between merging. And one of the last things I did was adding stuff for the ROC-RK3399-PC.

Otherwise I don't expect the chassis smbios changes and version identifier changes to cause issues. Though I did not test these exact changes on ROC-RK3399-PC yet, only on Pinebook (A64) and Pinebook Pro.

shvetsnikita, 2022-11-06:
> Do you have the exact commit ID?

No, unfortunately. I just downloaded zip to quickly build and test it.

samueldr, 2022-11-06:
Do you remember if it was after I pinged you about the changes for the ROC-RK3399-PC? If it was after, then there should have been no changes force-pushed that would affect this.

I guess this rests on me testing this on my device. Also maybe looking at Libre Computer's U-Boot (based on mainline) to see if they have any fixups for something.

shvetsnikita, 2022-11-06:
> Do you remember if it was after I pinged you about the changes for the ROC-RK3399-PC? 

I don't remember, sorry.

samueldr, 2022-11-07:
Oh, I think I might have assumed too much from context.

I assumed it crashed before attempting to boot an operating system. It looks like it crashes when booting an operating system.

This is a situation where #48 would have helped us both.

I had limited my testing, on the ROC-RK3399-PC, to starting Tow-Boot without an operating system available. Wherein I assumed if it booted successfully on other RK3399 systems, the same system would work fine on that build.

So, for anyone following at home, in my limited testing (still testing)

 - SPI-installed a5d2ea8ad9421ed70c736968535168fa712d982d will `"Synchronous Abort" handler, esr 0x96000010` with SD and USB NixOS UEFI iso (which was verified to work on Pinebook Pro).

samueldr, 2022-11-07:
After reproducing the abort, I was able to verify and fix it. #210 has the fix. It wasn't limited to the ROC-RK3399-PC. It would affect any system using SMBIOS (used by UEFI) and not defining a chassis-type in its U-Boot DT.

shvetsnikita, 2022-11-09:
> After reproducing the abort, I was able to verify and fix it. https://github.com/Tow-Boot/Tow-Boot/pull/210 has the fix. It wasn't limited to the ROC-RK3399-PC. It would affect any system using SMBIOS (used by UEFI) and not defining a chassis-type in its U-Boot DT.

Good to know! Regarding USB errors I mentioned earlier (_both throw USB-related errors in both cases, even if no USB devices is attached ("Failed to get clocks")_) - should I fill separate bugreport about it? Do you observe same errors during your attempts to boot NixOS iso?

samueldr, 2022-11-09:
USB works on my end when this error is printed.

I figure it's one of those we can safely ignore. What should be done is compare with mainline U-Boot and see if the same happens there. But we should probably upgrade to the latest U-Boot before comparing.

One hard-to-reconcile truth with some of the printing in U-Boot, is that it often happens that there are benign messages that can safely be ignored, but it's impossible to really know which.

samueldr, 2023-07-03:
Sorry it took a while to get back to this. I'm not 100% sure, but it looks like this ended-up being fixed. I'm assuming so and closing. If it isn't comment for me to re-open. (There's a release coming-up soon.)

closed 2023-07-03 by samueldr

shvetsnikita, 2023-08-01:
@samueldr, thank you for fixing this!

While USB mostly stable now for regular devices (keyboards, flash drives, etc.) there is still issues with anything more complicated. Below I describe results of testing with 2022.07 build from Release page flashed to SPI.

1. Anything USB3 capable is not detected in USB Type-C port. However simple USB1/USB2 devices attached via Type-A to Type-C adapter works.
2. Satechi Type-C Card Reader attached to USB Type-A (via adapter) or USB Type-C port, directly or via USB hub, cause backtrace and bootloop. 
3. Simple USB topology (like one Belkin F4U093 or one StarTech HB30C5A2CSC attached to USB Type-A port; not Type-C due to item 1) works fine - for example I can use keyboard attached to hub, boot from flash drive attached to hub, etc. However, if I put one hub into another - it become too much, and Tow-Boot can not initialize connection with whole setup, so if one hub attached to another, both hubs don't work.

samueldr, 2023-08-01:
What would be needed here is to confirm whether or not these *should* or should not work with U-Boot proper. All of the USB specifics, once it's started, are mainline U-Boot drivers and semantics.

shvetsnikita, 2023-08-02:
Re-tested USB support on upstream U-Boot. Is I said here USB doesn't work for me on U-Boot 2022.07, so this is not going to be apples to apples comparison, but here is what I got with U-Boot 2023.07:

1. Reproducible on U-Boot 2023.07
2. Not reproducible on U-Boot 2023.07
3. Not reproducible on U-Boot 2023.07. Even keyboard behind two sequentially attached hubs is detected and works.

PR #210 (2022-11-08): 2022.07/patches: Fix smbios chassis-type handling when missing
This actually would only work on these devices:

```
arch/arm/dts/sun50i-a64-pinebook.dts:   chassis-type = "laptop";
arch/arm/dts/sun50i-a64-pinetab.dts:    chassis-type = "tablet";
arch/arm/dts/sun50i-a64-pinephone.dtsi: chassis-type = "handset";
arch/arm/dts/sun50i-a64-teres-i.dts:    chassis-type = "laptop";
arch/arm/dts/rk3399-pinebook-pro.dts:   chassis-type = "laptop";
```

Since otherwise it would set the string to `NULL`, which apparently isn't well-liked by the implementation.

Note that this only affected Operating Syste [... cut]