Tow-Boot/Tow-Boot #212: PBP UART?

UltraBlackLinux, 2022-11-10:
Hey there,
coming from [this](https://forum.endeavouros.com/t/pinebook-pro-prebuilt-image/33676/25) topic, it seems like the non-spi version of tow-boot does not support [UART](https://wiki.pine64.org/wiki/Pinebook_Pro#Using_the_UART) - is that correct?

If not, how do you enable it? It's a really neat feature to have.

Thanks!

psstoyanov, 2022-11-18:
Any RK3399 device is usually using the following baud `1500000`. 
TwoBoot in comparison _is_ opinionated and has opted for `115200` as a default for all devices - even the PBP.
You can see my PR with some of the information: https://github.com/Tow-Boot/Tow-Boot/pull/182

Apart from that, it's good to check if your PBP has enabled the serial from the internal switch.

Glancing at the thread, there are a lot of mentions about having EndevourOS on the NVMe.
The peculiarities about that setup are separate from your question about TowBoot and UART.
In short if you want to have the NVMe as the only storage on the PBP, TowBoot has to be on the SPI.

I haven't looked at the setup EndevourOS is using but you can verify if the `extlinux.conf` for EndevourOS has the serial reporting to `115200` or `1500000`. 
iirc that was defined here for ManjaroARM if using uboot: https://gitlab.manjaro.org/manjaro-arm/packages/core/uboot-pinebookpro/-/blob/master/uboot-pinebookpro.install#L17
Not sure if that was removed from the Generic image setup on ManjaroARM - been awhile since I've looked deeply into those.

You will have to find the same config of EndevourOS yourself. If the EOS is using different baud for the tty defined, then you have to switch manually between the two bauds depending on the stage you are interested in.
Better yet - set it to be the same as the one used by TowBoot.

UltraBlackLinux, 2022-11-18:
> Apart from that, it's good to check if your PBP has enabled the serial from the internal switch.

I verified that the switch was set to enable UART

> TwoBoot in comparison is opinionated and has opted for 115200 as a default for all devices - even the PBP.

Even if I was using the wrong baudrate, I still would have seen garbled text, but that was not the case

I ended up flashing arch arm to an SD-Card and flashing EnOS from there. This worked perfectly fine. I haven't tested the UART since, but in case it does not work I will reopen this. Thanks for the info!

closed 2022-11-18 by UltraBlackLinux