Tow-Boot/Tow-Boot #215: Standardized baud rate is annoying

vincele, 2022-11-20:
Yes, this is clearly opinionated, in the opposite way of what you chose. ;-)

I'm on a pinebook pro, and the linux kernel often has the default rockchip 1.5Mbauds rate, so I only get messages either of tow-boot or the kernel.

But I'm OK with being annoyed.

What about putting a remainder on how to configure the kernel parameter accordingly: "You should also append `XXXXX` to your boot menu entry in extlinux.conf", for example.

I'd put that in :
https://tow-boot.org/serial.html
https://tow-boot.org/differences-from-u-boot.html

Argh, I tried appending `console=ttyS0,115200`, it is not working as expected...
So how can I have both tow-boot & kernel on the same rate ?

vincele, 2022-11-20:
Don't hit me, I put a smiley...

vincele, 2022-12-13:
Looks like https://github.com/Tow-Boot/Tow-Boot/pull/182 would fix this issue...

psstoyanov, 2022-12-13:
To me it sounds like a great addition.
I will see to make a small addition to #182 this Friday. 

I have seen a few variations on how the serial tty would behave depending on the distro config (Rcade/ Fedora/ Manjaro/ etc) that may not be 1:1 with the example. 

I would avoid going into detail of the combos that may affect it and add a note for the generic `exlinux.conf` that is found on most distros.

saper, 2023-05-07:
I also think all rockchip-based boards should use 1500000 bps. Not only because of Linux, but also if anything would come from Boot ROM before, it will be sent using this speed.

samueldr, 2023-07-03:
> I also think all rockchip-based boards should use 1500000 bps. Not only because of Linux, but also if anything would come from Boot ROM before, it will be sent using this speed.

The boot ROMs for all RK3[35]** systems I tested with prints nothing, ever, and any of the previous stages can be configured for 115200, when it's not the one built with U-Boot (which will be at the configured baud rate). Yes, this includes the proprietary ram training blobs. Changing the baud rate is *trivial enough*.

So this is not an issue.

samueldr, 2023-07-03:
Hi, sorry for the lengthy delay here (and well, this also applies to the previous comment).

> What about putting a remainder on how to configure the kernel parameter accordingly: "You should also append XXXXX to your boot menu entry in extlinux.conf", for example.

So, this is a really more nuanced problem. The baud rate used by Linux will depend on many different factors. One of those is that if there are no `console=` params, the "chosen" stdout will be derived from the device tree. In turn when using extlinux.conf, it will be common to load the device tree that the OS configures, instead of relying on the platform firmware innate device tree.

In the case of most, probably all, rockchip boards, it is configured like here:

 - https://github.com/torvalds/linux/blob/a901a3568fd26ca9c4a82d8bc5ed5b3ed844d451/arch/arm64/boot/dts/rockchip/rk3399-pinebook-pro.dts#L28-L30

The issue is multi-faceted here:

 - This means that the default console is not to the display, like commonly on ACPI UEFI boot.
 - This means that the baud rate is re-configured according to the opinion of another party (the kernel)

Luckily, the solution to the actual problem you are experiencing (mismatched expectation between parties involved in the OS boot) can be solved. The expected solution, the full solution relies on future updates to features, but the initial gist can be applied early:

 - When loading a DT, U-Boot already supports "fixups" to the DT (and is mandatory for stuff like mac addresses, serials, some hardware detection)
 - Use the fixup to delete the `/chosen/stdout` prop.

Later:

 - Provide a choice in the platform interface for a "DT chosen stdout", which would be "off", "serial", and maybe other device-specific choices where relevant.

Though this won't help when the distros make "device-specific" images and hardcodes kernel cmdline `console=` parameters. At that point it comes do a different mismatched expectation, and one that is "unsupported" here. We champion universal images. Board-specific images should only exist as a crutch for when hardware is unsupported by mainline. (But this is another topic entirely.)

Does that help?

vincele, 2023-07-03:
Yeah, any hint in the doc telling me, the baud rate you are using may come from the following sources: [...] and can be customized for your needs in the following files: [...] will help.

It's just to put people on the right track to fix their problems themselves.

samueldr, 2023-07-03:
(Now that it's confirmed I analyzed the problem correctly, I renamed the issue to better expose an actionable task.)