Tow-Boot/Tow-Boot #216: rEFInd will not load on two-boot on Pinebook Pro

petaramesh, 2022-11-22:
Hello, rEFInd https://www.rodsbooks.com/refind/ will not load on two-boot on Pinebook Pro.

I don't know if this might be a Tow-boot or a rEFInd issue though.

It displays (see attached picture) then immediately reboots.

Sorry for the poor quality of the pic, I had to get it with my phone, and id displays for so short that I didn't have a chance to get it any better...

[image]

samueldr, 2023-07-03:
Hi, sorry for the long time it took to reply.

I suspect this might be the issue found and fixed here:

 - https://sourceforge.net/p/refind/code/merge-requests/45/

If you can try with a recent release of rEFInd, that would be nice to confirm.

One problematic part with the abort handler is that it's hard to know who's at fault: the EFI runtime (U-Boot, Tow-Boot) or the program (here rEFInd).

petaramesh, 2023-07-05:
Hello,
Unfortunately, the situation was still the same with refind-0.14.0.2.

samueldr, 2023-07-05:
Hmm, so multiple angles:

 - I am running *a* rEFInd build on my Pinebook Pro successfully
 - There *was* a problem with rEFInd, and have been two "minor" [and two patch-level] releases since.
 - I am not running the prebuilt rEFInd from the sourceforge project, but the rEFInd as configured and built in NixOS. The version I'm running is 0.13.3

```
$ strings --encoding=b /boot/EFI/boot/bootaa64.efi | grep -i '0\.1'  
0.13.3
```

But since it's 99% sure it's configured differently enough from how the author builds their rEFInd build, I can't assume that 0.13.3 from the author should or shouldn't work.

Also I'm currently running the `2022.07-006-pre` Tow-Boot build, which could further make things different enough here.

I'll try and get back to this at some point, will figure out a good way to test this correctly (e.g. I'll have to test a matrix of [ 0.13 0.14 ] × [ author-built NixOS-built ]... but that will come after the upcoming release is cut.

samueldr, 2024-04-18:
Hi!

Is this still an issue on your end?

petaramesh, 2024-05-01:
Hi,

Sorry for the long delay, I hadn't had a chance to test it.

I'm happy to report that rEFInd 0.14.2 loads properly from Tow-Boot 2023.07-007 on my PineBook Pro :)

At least it displays a working menu, when it would just badly crash previously.

Information shows :

```
rEFInd Version 0.14.2

Running on :
    Firmware: Das U-Boot 8227.17
    EFI Revision 2.10
    Platform: ARM (64 bit); Secure Boot inactive (direct)
    Screen Output: Graphics Output (UEFI), 1920x1080
```

I haven't had a chance to actually boot an OS from it, as the OS currently installed on this machine is in plain legacy boot mode.

closed 2024-05-01 by petaramesh