Tow-Boot/Tow-Boot #217: [Pinebook Pro] [EFI] Insisting about DTBs provided in EFI mode

petaramesh, 2022-11-27:
I kept playing with the Pinebook Pro either in UEFI or in “extlinux legacy” boot modes using the same Manjaro with the same kernel version, and the performance difference is HUGE, going from “nice laptop” in syslinux mode to “barely usable experimental thing” in EFI mode.

In EFI mode, everything is much slower, block storage access crawls, the machine freezes for long periods everytime it needs to swap (even thought it's ZRAM swap...)

I checked the available memory in both EFI vs. legacy modes, and the memory overhead of the EFI system is rather neglectable.

So I thought of the DTBs again, and in both modes I extracted the DTBs from the running kernel using : 
dtc -I fs -O dts /sys/firmware/devicetree/base

So to check what both boot methods would actually provde the Linux kernel with

Then I saved the files and compared them using diff.

And I don't know much about analyzing DTS files, bug they are obviously *VERY* different in many places.

I wonder why DTS describing the same system hardware should show so big differences whether they are provided by an OS package or by the UEFI firmware.

Nevertheless, I can only confirm that my system flies starting from extlinux and using distribution-provided DTBs, and desperately crawls when using UEFI DTBs that come with Tow-boot.

I attach here obtained DTS files, should you want to take a look :

[PBP_tow-boot-2021.10-005_DTBs.zip](https://github.com/Tow-Boot/Tow-Boot/files/10098893/PBP_tow-boot-2021.10-005_DTBs.zip)

Strit, 2022-11-28:
The main difference is probably that the u-boot dts (which tow-boot also uses) was last synced from kernel 5.14 and the one you get from the kernel (and define in extlinux) is updated to match your kernel, so likely 6.0 at this point. That's a 2 year gap of improvements done by mainlining support.

So I'm not sure what you are asking here. U-boot has not synced the dts for a while and some dts stuff might not even be used in EFI mode, but rather the lacking (on ARM at least) ACPI tables.

That's my theory at least, but I'm not an expert.

petaramesh, 2022-11-28:
Hi,

Well, I undestood from my readings that the “preferred method” for booting ARM devices is now UEFI, and the Tow-boot website states that the vision of the project is that the system firmware should be OS-independant, as it is on PC archiitecture hardware, and not provided by a given OS / Linux distribution - freeing distribution maintainers from the task of providing software images specific to every possible ARM hardware out there.

So well, I wanted to give it a shot and play with it.

My results are clearly that : 
- It basically works
- But results in dramatic performance differences that makes the UEFI method currently practically unusable on a Pinebook Pro.

More detailed inspection shows that the way the hardware is described to the Linux kernel by both method is quite different.
Noticeably : 
- The eMMC device is called mmcblk2 withs distro provided DTBs and mmcblk0 by UEFI provided DTBs - whichi doesn't really matter but gives an evidence that there are differences.
- Diffing the DTBs gives a hell of a lot of output.

So my conclusion is that the UEFI method is not really usable as long as it doesn't give a system description that allows correct performance of the Linux kernel.

And “what I am asking here” is of course that this would be fixed to allow both methods to be equally usable, and allow using the UEFI “now preferred” method, that is more feature-complete and allows using a bootloader such as grub or rEFInd or any UEFI-capable one.

Otherwise I will (and Linux users will) have to stick to the legacy method on this machine.

Strit, 2022-11-28:
Sure, and the fix is for u-boot to sync the dts from kernel more often and then for tow-boot to follow upstream u-boot releases. But that are 2 giant "asks", where only the latter is a tow-boot issue.

petaramesh, 2022-11-28:
Well yes, but what would be the point of system firwmare projects that would lag 2 years behind for system description in the firmware they provide ?

vincele, 2022-11-29:
Maybe you can try to search for differences within linux.
Maybe by diffing kernel boot messages, or by having a look at:
- /sys/devices/system/cpu/cpu*/cpufreq/*
- /proc/interrupts

Something may stand out and help pinpoint the problem.