Tow-Boot/Tow-Boot #218: tag a new release with ppp 'rk_gpio: Added spl_gpio_input method' patch?

ollieparanoid, 2022-12-13:
Similar to #171, it would be great to have a new release tagged that includes: https://github.com/Tow-Boot/Tow-Boot/commit/c9a16b8cbf4823138eaae937cab467f2a908807a

I also realize that there are milestones and I looked into them, but not sure when the next release is planned. Maybe it makes sense to backport this patch to the previous release?

Thanks a lot!

samueldr, 2023-07-03:
Hi, sorry for the wait until *anything*.

Not sure if you've seen that there have been circumstances taking me away for the project for a while, and just now I'm taking the time to dust things off and prepare that next release.

So the plan for the short term is to make a release with the [last few trivial issues tackled (or verified)](https://github.com/Tow-Boot/Tow-Boot/milestones/2022.07-006). So while it's still a year out of date, the next things I'm tackling are first better improving things to ensure there's a reduced "bus factor".

Here's some more context around this, if you weren't aware:

 - https://gist.github.com/samueldr/8750e4aabbc23badec7a8798cabd4ed1

In the mid-term vision I'm looking to sync back up to the 2023.07 release, "as-is" (without new complex changes, without new features) and then see how development goes from here.

samueldr, 2023-07-06:
Closing, now that a release has been cut:

 - https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2022.07-006

closed 2023-07-06 by samueldr