Tow-Boot/Tow-Boot #222: Support for Radxa ROCK 4 SE (and ROCK 4 C plus)?

axelsimon, 2022-12-30:
I am not knowledgeable enough to understand if the variation in hardware (cf. [table](https://wiki.radxa.com/Rockpi4/hardware/models)) mean the ROCK 4 SE (and the ROCK 4 C plus?) need their own build of tow-boot or not, but i think they do.
I understand also that the warning regarding the lack of SPI flash from the [ROCK 4](https://github.com/Tow-Boot/Tow-Boot/tree/released/boards/radxa-RockPi4) is valid for these two boards, as they don't have SPI flash.

Thanks!

axelsimon, 2023-01-07:
For what it's worth, i was able to boot using the shared storage strategy with the latest Tow-Boot for ROCK 4 (`radxa-RockPi4-2021.10-005`) and a recent NixOS new kernel iso ( `nixos-minimal-new-kernel-23.05pre434398.33e0d99cbed-aarch64-linux.iso`).

I'm seeing messages at NixOS boot that don't look entirely positive, but again, not really knowledgeable enough to now what they mean.
Things like:
`spi-nor spi1.0: unrecognized JEDEC id bytes: ff ff fff ff ff ff`
`dw_wdt ff8480000.watchdog: No valid TOPs array specified`
`rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!`
and the one i think is preventing me from getting wifi to work:
`brcmfmac mmc2:9991: 1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rockpi4b.bin failed with error -2`

 I realise these are kernel messages, so i'm not sure if they are signs of the that Linux 6.0.12 doesn't have support for this board, or that Tow-Boot hasn't initialised everything as needed. Hopefully this helps a bit.

Strit, 2023-01-11:
Those are kernel messages yes.

The last one just seems to indicate that you are missing the firmware blob for the wifi. That's for your OS to provide.

axelsimon, 2023-01-26:
Thanks for your help, @Strit. According to Jack from Radxa:
> Build mainline u-boot is the same for all arm64 Rockchip soc. You can refer: http://opensource.rock-chips.com/wiki_U-Boot

I'm not sure if that explains why i was able to boot the ROCK 4 SE with the ROCK (Pi) 4 Tow-Boot image. I mean, if it was as simple as every arm64 Rockchip SoC needing the same U-boot, it seems that there would be no point in having multiple versions of Tow-Boot… I feel i'm missing something :)

Strit, 2023-01-27:
They are mostly the same, but there can be slight hardware changes between the boards, which is often reflected by different `defconfig` being used to build the different u-boots.

The SoC is the same, but the other hardware on the board might not be. Like the power regulator, which if the wrong one is assumed by u-boot/tow-boot, can burn out your board.

axelsimon, 2025-10-01:
It looks like Radxa have redone their wiki and put up a lot of useful hardware info, including schematics: https://docs.radxa.com/en/rock4/rock4ab-se/download

And from what i pieced together from Radxa resources, the Rock4SE uses the same bootloader as the Rock4B. Armbian also [say](https://www.armbian.com/rockpi4/):
> For Model SE, use the Model B builds [here](https://github.com/armbian/community/releases/latest).