Tow-Boot/Tow-Boot #223: Pinebook Pro - When booting from Tow-Boot on SPI, Linux Kernel can not see the eMMC drive.

ben-kuhn, 2023-01-12:
When booting using Tow-Boot, uBoot can see the EMMC drive and attempt to boot from it, but the kernel panics because it is unable to see the eMMC disk.  I have tested with Tow-Boot on an SD card and SPI.

Booting an OS from SD or NVME works fine.  /dev/mmcblk0 is not available booting from those devices.

I can reproduce this with two different eMMC modules so it does not appear to be a hardware issue with the storage.

Booting and accessing eMMC works fine using other uBoot implementations.

I can reproduce this with any distribution I try (NixOS, Armbian, Manjaro, etc...) with kernel versions from 5.15 to 6.2.

I can reproduce this with the last 3 Tow-Boot releases.  I have not tried releases older than that.

Strit, 2023-01-13:
Very odd. I don't have this issue.

Currently booting Manjaro from eMMC with Tow-Boot 005.

ben-kuhn, 2023-01-13:
I am happy to collect any logs, swap around hardware, or try just about anything.  Tow-Boot seems to work well for everyone else so this is probably a corner case that will need more information.

catbabylon, 2023-03-11:
I've noticed this issue as well. I'm currently running Manjaro generic EFI installed on the nvme (which is drawing 3.8W according to nvme-cli) and I noticed that the emmc doesn't show as a device when running lsblk, same error messages in dmesg logs.

I've taken the emmc out and will flash it via the usbflasher and see what happens when I try to boot directly from it via the towboot menu.

repagh, 2023-04-25:
Same problem here, If I boot by selecting the SD from the menu (and don't let tow-boot touch/scan the eMMC), I can access the eMMC from the OS on the SD. Once tow-boot accessed the eMMC, it can no longer be read from the OS on the SD.

JMasak, 2023-04-27:
I have the same problem with different distributions, using TowBoot 2021.10.-005.
On some rare occasions the system boots up from eMMC but most of the time i just get a lot of error messages. 
When i boot up the system from SD i can access the eMMC content without any problem.
I am using Manjaro Arm 23.02 (Generic EFI) on the SD card. 

Maybe someone who can boot from eMMC successfully can boot Manjaro from the SD card and rthen eboot the system from eMMC. If this then fails the problem could be cause maybe by the deinitialization from Manjaro during poweroff and a possible missing initialization step in TowBoot?

nagalun, 2023-10-28:
Oh I think I finally figured out the issue... Just took a bit of hair loss:

Reverting back to extlinux boot didn't seem to work for me either (even when having /boot on the SD, so Tow-Boot poked the eMMC as little as possible) and it was very difficult (10+ tries) to get the PBP to boot, but then I realized that the DTB is not being provided explicitly with the generated extlinux config (I guess on EFI is similar, it uses a Tow-Boot provided DTB?).

I added the `FDT` entry and incredibly enough, now I get consistent boots first try even though Tow-Boot prints `Card did not respond to voltage select! : -110` when trying to read eMMC. I'm using 2023-007-rc3.

I guess the solution would be to upgrade the integrated DTB with a more recent one? I looked a bit around the repo but couldn't find how to provide my own, https://github.com/Tow-Boot/Tow-Boot/issues/217 seems relevant and from what I understood I'd have to touch the u-boot sources.