Tow-Boot/Tow-Boot #227: Odroid N2+ spi installer menu doesn't respond / locks up

dubschro, 2023-02-10:
Odroid N2+ 4B odroid-N2-2021.10-005 gets to the boot menu with spi installer but the menu doesn't draw the lower right corner and the keyboard will not move the menu around.  It's like it freezes when drawing the last line of the menu container box.  The logo appears at the bottom right of the screen, but it has no response to keyboard input.  Not sure how to assist.  Will try older versions to see if it was introduced in this one.

dubschro, 2023-02-10:
Issue is in previous release as well.

bradennapier, 2023-07-19:
Also running into this one.

dubschro, 2023-11-23:
Same exact behavior with odroid-N2-2022.07-006.tar.xz.

dubschro, 2024-01-23:
> I have one Odroid-N2 and a dozen Odroid-N2+ boards. I haven't seen this
> issue on any of them however I generally use the serial console to interact
> with Tow-Boot. Are you using the HDMI output and USB ports for the keyboard?

Yes, HDMI and USB.

angerman, 2024-09-26:
Just hit this today as well. I believe I've also seen this on the rockpi4 at some point.

CtrlC-Root, 2025-01-23:
I just ran into a similar issue but on a Libre Computer Le Potato ([tow-boot](https://tow-boot.org/devices/libreComputer-amlS905xAc.html) and [manufacturer](https://libre.computer/products/aml-s905x-cc/)). This time I have an HDMI capture card, a USB keyboard, and the serial port connected at the same time. The USB keyboard is powered on but the screen displayed over HDMI doesn't update no matter what I do. Once I start using the serial console though the screen updates but seemingly about one action behind where I am in the serial console. So if I am on the menu screen and I use the down arrow to navigate through the options then the HDMI output shows the previously selected entry while the serial console shows the currently selected entry. Check out this screenshot of the HDMI capture on top of the serial console:

[image]

Another example from the menu:

[image]

Notice how the last line of output is missing. When I initially boot the system this is what the screen looks like until I interact with the menu through the serial console:

[image]

This artifact is pretty consistent. I'm going to hook up one of my Odroid N2+ boards and see if I can reproduce this behavior while I have both the HDMI and serial console connected.