Tow-Boot/Tow-Boot #230: PinePhone A64: Can't enter Jumpdrive mode

sven-frotscher, 2023-03-17:
I'm following Mobian's [install instructions](https://wiki.mobian-project.org/doku.php?id=install-linux) for the PinePhone and getting stuck at booting the phone into Jumpdrive mode.
As requested on the Mobian wiki page, I installed Tow-Boot on my PinePhone by following [these instructions](https://tow-boot.org/devices/pine64-pinephoneA64.html).

The install seems to have worked, booting without a USB connection or microSD card directly after the install shows this:

<details><summary>Serial terminal log</summary>
<p>

```

U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
DRAM: 3072 MiB
Trying to boot from MMC2
NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x209cf40, model: Pine64 PinePhone (1.2)

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)005 [variant: mmcboot]

CPU:   Allwinner A64 (SUN50I)
Model: Pine64 PinePhone (1.2)
DRAM:  3 GiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.
off
off
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:2...
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
MMC: no card present
Scanning disk mmc@1c0f000.blk...
Disk mmc@1c0f000.blk not ready
Card did not respond to voltage select! : -110
Scanning disk mmc@1c10000.blk...
Disk mmc@1c10000.blk not ready
Scanning disk mmc@1c11000.blk...
Found 2 disks
** Unable to read file ubootefi.var **
Failed to load EFI variables
BootOrder not defined
EFI boot manager: Cannot load any image
MMC: no card present
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
No ethernet found.
No ethernet found.

ERROR: Could not boot anything from distro_bootcmd.

poweroff ...
```

</p>
</details>
Before <code>poweroff ...</code>, the LED flashes red and the phone vibrates 10 times.

However, when I continue to follow along with the instructions in the Mobian wiki (Option 2: Mounting the target device's eMMC using TowBoot) by holding Volume Up while connecting the phone to my computer using a known good USB-A-to-C cable and it behaves almost identically (only the two lines below the `boot menu` message in the log have changed and the LED briefly lights up blue during these lines):

<details><summary>Serial terminal log</summary>
<p>

```

U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
DRAM: 3072 MiB
Trying to boot from MMC2
NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x209cf40, model: Pine64 PinePhone (1.2)

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)005 [variant: mmcboot]

CPU:   Allwinner A64 (SUN50I)
Model: Pine64 PinePhone (1.2)
DRAM:  3 GiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.
on
** Invalid partition 1 **
off
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:2...
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
MMC: no card present
Scanning disk mmc@1c0f000.blk...
Disk mmc@1c0f000.blk not ready
Card did not respond to voltage select! : -110
Scanning disk mmc@1c10000.blk...
Disk mmc@1c10000.blk not ready
Scanning disk mmc@1c11000.blk...
Found 2 disks
** Unable to read file ubootefi.var **
Failed to load EFI variables
BootOrder not defined
EFI boot manager: Cannot load any image
MMC: no card present
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
No ethernet found.
No ethernet found.

ERROR: Could not boot anything from distro_bootcmd.

poweroff ...

```

</p>
</details>

As you can see, holding Volume Up accomplished almost nothing.

So, where did I go wrong? Did I brick my PinePhone by flashing the eMMC too often or is there a different issue at play here? I'm also not opposed to diagnosing this further, but someone would have to walk me through that.

gitcoo, 2023-10-04:
Hi @sven-frotscher 

I believe I have the same exact situation.

I can install tow-boot to the emmc from balena, when I try to reboot with the volume up / power button, it buzzes and flashes red and eventually gives up.

Were you able to get the device to boot mobian or any os off the emmc?

Thanks!

sven-frotscher, 2023-10-05:
I wasn't able to boot anything except the Manjaro Plasma which my PP came with (not even a re-image of that) from eMMC yet.
Booting from microSD worked for megi's multi-boot test image (multiple distros and a working Jumpdrive) and the Tow-Boot installer, which sadly put my PP in this no-booting state.
I may give Jumpdrive from the multi-boot image another try to erase my eMMC or burn another OS/bootloader image on it when I have some more time and motivation on my hands.

If you have a spare 3.5mm TRS/TRRS plug and a USB-serial converter, you can DIY a serial console cable, flip the audio killswitch, and get some logs out of the various bootloaders, like I did. Maybe something in there will help the devs help us with our issue, who knows.

gitcoo, 2023-10-06:
Thanks @sven-frotscher !

I will look on ebay for the 3.5mm TRS/TRRS plug and a USB-serial converter, definitely like having more logs around to look into!

I did manage to get mobian back on my Pinephone A64 emmc, by using the image pointed to here:

https://forum.pine64.org/showthread.php?pid=116155#pid116155
https://github.com/dreemurrs-embedded/Jumpdrive/releases/download/0.8/pine64-pinephone.img.xz

I flashed the above image with balena etcher to an sd card and then booted with the sd card into Jumpdrive.

With Jumpdrive on screen, I was able to flash the above image to emmc with balena etcher. Then it booted (no sd card) just fine from emmc.

However, I think I still need to install tow-boot back onto the emmc to have a 'proper' installation, if I understand correctly.

sven-frotscher, 2023-10-06:
Yes, 3.5mm plug into the headphone jack, 3 wires (GND, Rx, Tx) to converter, converter's USB to computer. See [the PP wiki article](https://wiki.pine64.org/wiki/PinePhone#Serial_console) for details and for links to some ready-made console cables, if DIY/soldering isn't your thing.

Did I read that forum post correctly: I "just" need to boot Jumpdrive from a microSD and use that to flash Mobian onto the eMMC? Won't that overwrite Tow-Boot? Or do I then need to reinstall Tow-Boot onto eMMC using [its installer](https://tow-boot.org/devices/pine64-pinephoneA64.html)?

gitcoo, 2023-10-06:
Ok great thanks for the info, 

I just now flashed the tow-boot mmcboot.installer.img to sd, booted and with the tow-boot interface, "installed tow-boot" to emmc.

When I rebooted without power + up/down, it continued normally to the mobian that I already had on emmc.

When I try to boot with power button + vol up, it stops and has a blank screen with the blue led lit, for at least 10 minutes, so I gave up. I don't know if it should have booted to tow-boot or not?

I rebooted normally and it loaded into mobian again.

So all said, mobian continues to run fine off emmc, and I don't know if tow-boot is on the mmc boot record.

When I boot it plugged in via usb, I see rootfs in the file manager. I open balena etcher and see that the options for where to flash an image to show 

Linux UMS disk 0 (boot, rootfs) 31.GB /dev/sda

 and that is with no sd card in the PP.

So, I am a bit confused as to if I have tow-boot working or not.

sven-frotscher, 2023-10-06:
Tow-Boot goes into Jumpdrive mode if you hold Vol+ while booting, so you should see the eMMC as a disk (rootfs) when connecting it to your computer. Confusion could be minimized if it also showed something on screen in addition to turning on the blue LED.
Do you also see that disk when booting your PP with a USB connection but without touching the volume buttons?

gitcoo, 2023-10-06:
When the PP is off and I plug it into the laptop usb without holding down any button, it boots into mobian and I see the message on my laptop:

Connection 'Wired connection 2' activated

So then I power off (still plugged into usb).

Then I do the power + vol up to boot the device. IThe PP shows red led, then blue, I let go, and on my laptop I get the message rootfs with the option to mount and open (from dolphin file manager, I believe).

I do not know if that is tow-boot at work or if that would happen without having "installed" tow-boot as described above.

I do not know how to determine if tow-boot is installed properly.

sven-frotscher, 2023-10-06:
'Wired connection 2' seems to be a USB network connection, while rootfs is almost certainly Jumpdrive at work. Since your PP boots both Mobian and Jumpdrive successfully, I think you installed Tow-Boot into your eMMC correctly.

gitcoo, 2023-10-06:
Ok thanks,

I am a little confused, still.

I had mobian installed already, before I installed tow-boot, and I was only able to get mobian onto the PP with Jumpdrive intially. I was not able to install mobian before all of this, even though I had tow-boot on the emmc.

So, I have not successfully installed tow-boot on a blank emmc, and then installed mobian afterwards, so I will attempt that process, just to make sure I am not depending on Jumpdrive to have mobian installed.

Because I thought with tow-boot installed, you could power on with power + vol up, but that does not give me the tow-boot interface in the current state.

Thanks again for your suggestion and help!

sven-frotscher, 2023-10-06:
You flashed https://images.mobian.org/pinephone/mobian-pinephone-phosh-12.0.img.gz in order to install Mobian, correct?

gitcoo, 2023-10-06:
Yes, that was via the Jumpdrive image above on my emmc, not via tow-boot.

sven-frotscher, 2023-10-06:
Yes, that seems like the way to go for starting from scratch. Maybe you also need to zero out the beginning of the eMMC before installing Tow-Boot, I'm not sure.

sven-frotscher, 2023-10-06:
My PP boots Jumpdrive from microSD with a USB connection, but doesn't show up as a USB device of any kind on my computer. Connecting a serial cable somehow messes with the boot process (not even a Jumpdrive splash screen) and shows:
```
DRAM: 3072 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.2(release):v2.2-904-gf9ea3a629
NOTICE:  BL31: Built : 15:32:12, Apr  9 2020
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x4064410, model: PinePhone
NOTICE:  PSCI: System suspend is unavailable

U-Boot 2020.07 (Nov 08 2020 - 00:15:12 +0100)

DRAM:  3 GiB
MMC:   Device 'mmc@1c11000': seq 1 is in use by 'mmc@1c10000'
mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

starting USB...
No working controllers found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
653 bytes read in 3 ms (211.9 KiB/s)
## Executing script at 4fc00000
gpio: pin 114 (gpio 114) value is 1
4275261 bytes read in 190 ms (21.5 MiB/s)
Uncompressed size: 10170376 = 0x9B3008
36162 bytes read in 4 ms (8.6 MiB/s)
1078500 bytes read in 51 ms (20.2 MiB/s)
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Ramdisk to 49ef8000, end 49fff4e4 ... OK
   Loading Device Tree to 0000000049eec000, end 0000000049ef7d41 ... OK

Starting kernel ...
```
after which it just hangs.

sven-frotscher, 2023-10-06:
See dreemurrs-embedded/Jumpdrive#85 and dreemurrs-embedded/Jumpdrive#63 for the issue report, but no solutions (yet? One can hope...).

ArenM, 2023-10-07:
@gitcoo it sounds like tow-boot is working as expected. U-boot doesn't have a driver for the display, so the only interface you'll get is the blue led and a usb connection.

@sven-frotscher perhaps it would be possible to pinpoint your issue a bit more
 - If you boot from a sd card can you access the emmc? If so what is the partition layout?
 - When booted to linux are you able to make the phone appear as a usb device? Enabling usb networking is probably the most common use of this (or if you can telnet jumpdrive, but I'm not sure if that would get set up with the issues you're having there).
 - Can tow-boot access your emmc? With the serial console, you should be able to press escape to access a boot menu and select the firmware console, from there the command `ls mmc 1` should list the files on the first partition of the emmc.

gitcoo, 2023-10-07:
@ArenM 

Big thanks for the confirmation, it's onward from here then! ;0)

sven-frotscher, 2023-10-07:
>     * If you boot from a sd card can you access the emmc? If so what is the partition layout?

Booting the Jumpdrive microSD (disk image linked above), I get a UART shell if I don't just use my USB-Serial converter. This gives me the following info:
```
/ # ls -l sys/dev/block/ | grep -v loop
lrwxrwxrwx    1 0        0                0 Jan  1 00:00 179:0 -> ../../devices/platform/soc/1c0f000.mmc/mmc_host/mmc0/mmc0:59b4/block/mmcblk0
lrwxrwxrwx    1 0        0                0 Jan  1 00:00 179:1 -> ../../devices/platform/soc/1c0f000.mmc/mmc_host/mmc0/mmc0:59b4/block/mmcblk0/mmcblk0p1
lrwxrwxrwx    1 0        0                0 Jan  1 00:00 179:32 -> ../../devices/platform/soc/1c11000.mmc/mmc_host/mmc2/mmc2:0001/block/mmcblk2
lrwxrwxrwx    1 0        0                0 Jan  1 00:00 179:33 -> ../../devices/platform/soc/1c11000.mmc/mmc_host/mmc2/mmc2:0001/block/mmcblk2/mmcblk2p1
lrwxrwxrwx    1 0        0                0 Jan  1 00:00 179:64 -> ../../devices/platform/soc/1c11000.mmc/mmc_host/mmc2/mmc2:0001/block/mmcblk2/mmcblk2boot0
lrwxrwxrwx    1 0        0                0 Jan  1 00:00 179:96 -> ../../devices/platform/soc/1c11000.mmc/mmc_host/mmc2/mmc2:0001/block/mmcblk2/mmcblk2boot1
/ # fdisk -l
Disk /dev/mmcblk0: 7523 MB, 7888437248 bytes, 15407104 sectors
120368 cylinders, 4 heads, 32 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1 *  16,0,1      799,3,32          2048     102399     100352 49.0M  c Win95 FAT32 (LBA)
Disk /dev/mmcblk2: 14 GB, 15551660032 bytes, 30374336 sectors
1890 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk2p1    0,1,1       1023,254,63        104   30374311   30374208 14.4G  b Win95 FAT32
```
`mmcblk0` seems to be the 8GB microSD I installed Jumpdrive on.

sven-frotscher, 2023-10-07:
> Can tow-boot access your emmc? With the serial console, you should be able to press escape to access a boot menu and select the firmware console, from there the command `ls mmc 1` should list the files on the first partition of the emmc.

I tried that with the eMMC Tow-Boot (crazily mashing Escape):
```

U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
DRAM: 3072 MiB
Trying to boot from MMC2

NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x209cf40, model: Pine64 PinePhone (1.2)

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)005 [variant: mmcboot]

CPU:   Allwinner A64 (SUN50I)
Model: Pine64 PinePhone (1.2)
DRAM:  3 GiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.
Creating menu...
```
[menu, I select Firmware Console]
```
Use `run menucmd` to return to the menu.

=> ls mmc 1
** Invalid partition 1 **
Couldn't find partition mmc 1
```
What else can I try in that console?

ArenM, 2023-10-07:
The view from jumpdrive looks alright, /dev/mmcblk2 is the emmc. Did you check if you were able to access it over telnet to make sure the usb port is functional?

Is mmcblk2p1 formatted fat32, or does it just have the flag set on it? I'm not totally sure if that error means u-boot doesn't know about the emmc, or just the partition. You could also try running `mmc dev 1` and then `mmc info` as another method to see if u-boot can access the emmc.

sven-frotscher, 2023-10-07:
> The view from jumpdrive looks alright, /dev/mmcblk2 is the emmc. Did you check if you were able to access it over telnet to make sure the usb port is functional?

The Jumpdrive-only image from dreemurrs' repo doesn't want to make a USB connection to my PC, but the one included in megi's multi-boot image does (network connection and block devices for eMMC and microSD). I'll try flashing the eMMC using that.

> Is mmcblk2p1 formatted fat32, or does it just have the flag set on it? I'm not totally sure if that error means u-boot doesn't know about the emmc, or just the partition. You could also try running `mmc dev 1` and then `mmc info` as another method to see if u-boot can access the emmc.

```
=> mmc dev 1
switch to partitions #0, OK
mmc1(part 0) is current device
=> mmc info
Device: mmc@1c11000
Manufacturer ID: 12
OEM: 134 
Name: ASTCX.
Bus Speed: 52000000
Mode: MMC High Speed (52MHz)
Rd Block Len: 512 
MMC version 5.1 
High Capacity: Yes 
Capacity: 29 GiB 
Bus Width: 4-bit
Erase Group Size: 512 KiB 
HC WP Group Size: 8 MiB 
User Capacity: 29 GiB WRREL
Boot Capacity: 4 MiB ENH 
RPMB Capacity: 4 MiB ENH 
Boot area 0 is not write protected
Boot area 1 is not write protected
=> ls mmc 1
** Invalid partition 1 **
Couldn't find partition mmc 1
=> 
```
That seems to work at least.

sven-frotscher, 2023-10-07:
Looking at the partition labels and GParted, I must have flashed an x86 bootable ISO onto the eMMC somehow. No wonder every aarch64 bootloader under the sun complains about that :sweat_smile:

Wish me luck with flashing Mobian :)

sven-frotscher, 2023-10-07:
Mobian seems to boot, but not much more:
```

U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
DRAM: 3072 MiB
Trying to boot from MMC2
NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x209cf40, model: Pine64 PinePhone (1.2)

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)005 [variant: mmcboot]

CPU:   Allwinner A64 (SUN50I)
Model: Pine64 PinePhone (1.2)
DRAM:  3 GiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.
off
off
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
526 bytes read in 1 ms (513.7 KiB/s)
1:	Mobian GNU/Linux 6.1-sunxi64
Retrieving file: /initrd.img-6.1-sunxi64
18547242 bytes read in 399 ms (44.3 MiB/s)
Retrieving file: /vmlinuz-6.1-sunxi64
7763893 bytes read in 168 ms (44.1 MiB/s)
append: root=UUID=e72b69d1-5326-4d15-b9e8-cefe391a1e97 consoleblank=0 loglevel=7 ro quiet splash plymouth.ignore-serial-consoles vt.global_cursor_default=0
Retrieving file: /dtb-6.1-sunxi64/allwinner/sun50i-a64-pinephone-1.2.dtb
40966 bytes read in 5 ms (7.8 MiB/s)
   Uncompressing Kernel Image
Moving Image from 0x40080000 to 0x40200000, end=41670000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Ramdisk to 48e4f000, end 49fff22a ... OK
   Loading Device Tree to 0000000048e41000, end 0000000048e4e005 ... OK

Starting kernel ...

[    1.533698] lima 1c40000.gpu: error -ENODEV: _opp_set_regulators: no regulator (mali) found
[    2.395292] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    2.422248] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    2.431733] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    2.441961] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    2.451621] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    2.570123] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    2.581823] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    3.273361] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    3.351405] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    3.369731] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    3.435468] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    3.638472] panel-sitronix-st7703 1ca0000.dsi.0: mipi_dsi_attach failed (-517). Is host ready?
[    3.684848] pinephone-keyboard 2-0015: Failed to read device ID: -6
[    9.701127] request_module: modprobe binfmt-e687 cannot be processed, kmod busy with 50 threads for more than 5 seconds now
[    9.712285] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
[    9.731088] CPU: 0 PID: 1 Comm: run-init Tainted: G            E      6.1-sunxi64 #1
[    9.731099] Hardware name: Pine64 PinePhone (1.2) (DT)
[    9.731104] Call trace:
[    9.746409]  dump_backtrace+0xe8/0x11c
[    9.750176]  show_stack+0x1c/0x30
[    9.753495]  dump_stack_lvl+0x60/0x78
[    9.753507]  dump_stack+0x14/0x2c
[    9.753515]  panic+0x160/0x338
[    9.753525]  make_task_dead+0x0/0x170
[    9.753536]  do_group_exit+0x38/0xa0
[    9.753545]  __arm64_sys_exit_group+0x1c/0x20
[    9.753554]  invoke_syscall+0x4c/0x110
[    9.753567]  el0_svc_common.constprop.0+0x48/0xf0
[    9.753579]  do_el0_svc+0x30/0xb4
[    9.753590]  el0_svc+0x14/0x4c
[    9.753598]  el0t_64_sync_handler+0xec/0x120
[    9.753608]  el0t_64_sync+0x14c/0x150
[    9.753621] SMP: stopping secondary CPUs
[    9.757167] Kernel Offset: disabled
[    9.757170] CPU features: 0x00000,00c00000,0000401b
[    9.757177] Memory Limit: none
```
The Mobian splash screen shows up around the `panel-sitronix` lines and disappears later.
After the above messages, U-Boot starts again, repeating the process.

At least it is now a Mobian issue, not a Tow-Boot one anymore (I think).

sven-frotscher, 2023-10-09:
Update: The kernel panic only happens with https://images.mobian.org/pinephone/mobian-pinephone-phosh-12.0.img.gz, the [current weekly image](https://images.mobian.org/pinephone/weekly/mobian-pinephone-phosh-20231008.img.xz) boots and runs fine. Thank you for your help, @ArenM!

sven-frotscher, 2023-10-09:
The question of no USB ethernet/block devices when powering on Tow-Boot with Vol+ or when using the Jumpdrive image standalone is still open. What does megi's multiboot/p-boot do differently to make Jumpdrive work for me?