Tow-Boot/Tow-Boot #235: No HDMI audio on the RockPro64

alscaldas, 2023-06-01:
Hello,

I've just flashed my RP64 spi with Two-boot and it's mostly working but from one point: I get no HDMI audio. I usually connect my RP64 to the TV using HDMI and I'm using LibreELEC via PXE (although it doesn't work when I flash it to a SD either). I see all the audio options but none of them works. By the way, if I just plug a headphone on the RP64 audio output I got audio, which indicates that the issue is related to the HDMI.

```
LibreELEC:~ # aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDMI], device 0: ff8a0000.i2s-i2s-hifi i2s-hifi-0 [ff8a0000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Analog [Analog], device 0: ff890000.i2s-ES8316 HiFi ES8316 HiFi-0 [ff890000.i2s-ES8316 HiFi ES8316 HiFi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

There are no apparent error msgs when booting and I must say that the same issue also happens when using u-boot.

Device: RockPro64 v 2.1
Two-boot: 2021.10-005
LibreELEC: 11.0.1

Any help would be much appreciated.

CtrlC-Root, 2023-10-01:
Hi, I also flashed Tow Boot to the SPI chip on my RockPro64, and it seems to work as I can access the boot menu over the serial port. I don't have any HDMI output from the bootloader at all. Do you see the graphical Tow Boot menu on your board?

alscaldas, 2023-10-02:
Hey, thanks for taking the time to have a look at my issue. 

In fact it still happens. I do see the menu and can choose a boot option (another annoying issue, btw... to have my external USB connected disk properly recognized I always have to disconnect it, connect it again and then rescan the USBs when booting. But I believe it may be a matter of proper power, since it works if booting from a USB stick connected to the same USB port).

Anyway, LibreELEC boots fine and everything works except for HDMI audio.