Tow-Boot/Tow-Boot #249: Odroid-N2+ boot menu devices are off-by-one

mth, 2023-07-09:
Using the latest 2022.07-006 tow-boot release on Odroid-N2+.

The eMMC menu item tries to boot non-existent device 2 and SD menu item a device 1 which is actually eMMC (as the SD card is device 0). That is, both menu items device number is real_device_number + 1.

I guess that something has gone wrong with the predictable boot order changes.

Adding pictures from TV screen, if needed I can also attach serial console.

[image]
[image]
[image]

Selecting eMMC gives -5 error due some unrelated issue. Debian booted from USB using UEFI had same problem when accessing eMMC, but Armbian using its own device tree sees the Foresee HS200 eMMC module fine - I'm suspecting some problem with uboot device tree file when using this module.

mth, 2023-07-09:
I fixed it for myself by editing the bootcmd_mmc1 environment variable to use devnum=1 and saving the environment to the spi flash, same could be done with bootcmd_mmc2 for eMMC, so there is workaround for the bug using existing images.

samueldr, 2023-07-10:
Thanks, I'll have to dig and see if something changed in attributing the indexes for amlogic outright, or if it's specific to N2+

tpwrules, 2023-07-10:
I am running into the same problem on an ODROID C4 which is also Amlogic. Unfortunately there's no flash on that board to save an environment to so I have to downgrade.

As described, manually invoking `devnum=0; run mmc_boot` works fine to boot the SD card.

samueldr, 2023-07-10:
> my mmc list numbers are the same on -005 and -006 and i have three mmc targets called SD=0, eMMC=1, and (unused)=2.

tpwrules, 2023-07-10:
With my ODROID-C4, on -005:
```
=> mmc list
sd@ffe05000: 0 (SD)
mmc@ffe07000: 1

=> env print
[...]
target_name_mmc0=SD
target_name_mmc1=eMMC
target_name_mmc2=(Unused)
[...]
```

And on -006:
```
=> mmc list
sd@ffe05000: 0
mmc@ffe07000: 1

=> env print
[...]
target_name_mmc0=(sdio)
target_name_mmc1=SD
target_name_mmc2=eMMC
[...]
```

tpwrules, 2023-07-10:
I get identical outputs and results on my ODROID-N2 (non plus) with Tow-Boot installed to SPI.

samueldr, 2023-07-11:
Thanks, that pretty much confirms my two mistakes, compounding:

 - Accidentally not keeping mmc0 in the boot order
 - Messing with the names

CtrlC-Root, 2023-10-02:
> I fixed it for myself by editing the bootcmd_mmc1 environment variable to use devnum=0 and saving the environment to the spi flash. Same could be done with bootcmd_mmc2 for eMMC, so there is a workaround for the bug using existing images.

This worked for me as well and I can now boot off the SD card and eMMC correctly on the Odroid-N2 (non-plus) variant.

PR #274 (2023-10-16): Mark 007-rc2
Solves issues with Amlogic (meson) boot order and MMC naming.

See:

 - `git diff tb-2023.07-007-rc1..tb-2023.07-007-rc2`
 - https://github.com/Tow-Boot/U-Boot/compare/tb-2023.07-007-rc1..tb-2023.07-007-rc2

<details><summary>Old PR body</summary>
As I briefly touched on elsewhere, all my amlogic boards are *weird*.

I'd really like someone to verify it fixes the predictable boot order issue.

**IMPORTANT NOTE**: If you have saved an environment to SPI flash, you will need to wipe the whole SPI flash (or the env, if you know how to), as otherwise your test will be invalid.

 - Clo [... cut]
closed 2023-10-16 by samueldr