Tow-Boot/Tow-Boot #258: Build for RK3399 targets without SPI fail

shvetsnikita, 2023-08-01:
Hello!

While Tow-Boot ports to AllWinner H3/H5-based boards (such as [NanoPi M1](https://github.com/Tow-Boot/Tow-Boot/pull/257) and [Libre Computer ALL-H3-CC H5](https://github.com/Tow-Boot/Tow-Boot/pull/256)) cold be quickly drafted and built, this is not the case with RK3399-based boards without SPI (such as NanoPC-T4 and NanoPi R4S) as build fail with following error message:

```
Warning: CONFIG_SPL_ENV_IS_NOWHERE is expected to be set to «CONFIG_SPL_ENV_IS_NOWHERE=y», but is not present in config file.
ERROR: CONFIG_SPL_SPI_FLASH_SFDP_SUPPORT is expected to be set to «CONFIG_SPL_SPI_FLASH_SFDP_SUPPORT=y», but is not present in config file.
ERROR: CONFIG_SYS_SPI_U_BOOT_OFFS is expected to be set to «CONFIG_SYS_SPI_U_BOOT_OFFS=0x80000», but is not present in config file.
Warning: CONFIG_TPL_ENV_IS_NOWHERE is expected to be set to «CONFIG_TPL_ENV_IS_NOWHERE=y», but is not present in config file.
```

samueldr, 2023-08-01:
Hi!

Just identified the core reason. Can you test with this change?

```
diff --git a/modules/hardware/rockchip/default.nix b/modules/hardware/rockchip/default.nix
index 973cfb5..b578093 100644
--- a/modules/hardware/rockchip/default.nix
+++ b/modules/hardware/rockchip/default.nix
@@ -16,6 +16,7 @@ let
     variant
   ;
   cfg = config.hardware.socs;
+  withSPI = config.hardware.SPISize != null;
 
   firmwareMaxSize = 4 * 1024 * 1024; # MiB in bytes
   partitionOffset = 64; # in sectors
@@ -46,7 +47,7 @@ in
     (mkIf cfg.rockchip-rk3399.enable {
       system.system = "aarch64-linux";
       Tow-Boot = {
-        config = [
+        config = mkIf withSPI [
           (helpers: with helpers; {
             # SPI boot Support
             MTD = yes;
```

I tested against `nanopc-t4-rk3399_defconfig` to reproduce, and then after the fix it builds.

samueldr, 2023-08-09:
Okay, but the core issue here (targets without SPI can't be built) is solved, right? I would like issues to be solveable. In other words, actionable "one thing only" issues, with additional issues for other problems.

(It won't be closed until that change is merged into Tow-Boot.)

* * *

The *other* issues I suspect come from upstream U-Boot.

Note that armbian loves to ship older versions of things... here it's possibly for `v2022.04` considering the directory name. And sometimes they ship vendor forks of U-Boot, rather than mainline U-Boot. So comparing with armbian is *really hard* without a deep dive in their build framework to suss out the exact build details.

Can you open a new issue with details and observations for that specific board, so that it is more easily tracked and digested? Can you also try against mainline U-Boot 2023.07, so it's made more obvious whether or not it's a Tow-Boot issue or not?

shvetsnikita, 2023-08-10:
> Okay, but the core issue here (targets without SPI can't be built) is solved, right?

Yes. Once again, thank you for solving this!

> Can you open a new issue with details and observations for that specific board, so that it is more easily tracked and digested?

Sure: https://github.com/Tow-Boot/Tow-Boot/issues/260

samueldr, 2023-10-18:
Fixed in

 - https://github.com/Tow-Boot/Tow-Boot/commit/2b39fec5111409fb311f8ff4eeec55668cc50737
 - https://github.com/Tow-Boot/Tow-Boot/commit/2b39fec5111409fb311f8ff4eeec55668cc50737

closed 2023-10-18 by samueldr