Tow-Boot/Tow-Boot #260: FriendlyElec NanoPC-T4 support

shvetsnikita, 2023-08-10:
Moved from [here](https://github.com/Tow-Boot/Tow-Boot/issues/258#issuecomment-1671651134).

Issues I found so far:
1. Even with [these build options](https://github.com/armbian/build/blob/main/patch/u-boot/u-boot-rockchip64/u-boot-rk-rk3399-usb-start-nanopc-t4.patch) USB doesn't work on Tow-Boot 2022.07 and 2023.07:
```
=> usb start
starting USB...
Bus usb@fe380000: ehci_generic usb@fe380000: Failed to get clocks (ret=-19)
Port not available.
Bus usb@fe3a0000: ohci_generic usb@fe3a0000: failed to get usb phy
Port not available.
Bus usb@fe3c0000: ehci_generic usb@fe3c0000: Failed to get clocks (ret=-19)
Port not available.
Bus usb@fe3e0000: ohci_generic usb@fe3e0000: failed to get usb phy
Port not available.
Bus usb@fe800000: Failed to get PHY0 for usb@fe800000
Port not available.
Bus usb@fe900000: Failed to get PHY0 for usb@fe900000
Port not available.
```
Yet somehow USB works in Armbian's U-Boot 2022.04 build:
```
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe900000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found
scanning bus usb@fe900000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
```
I probably need something else from [this folder](https://github.com/armbian/build/tree/main/patch/u-boot/u-boot-rockchip64-v2022.04), but I have no idea what it could be. Any advice?

2. Ethernet on Tow-Boot 2022.07 is like this:
```
Net:   
Error: ethernet@fe300000 address not set.
No ethernet found.
```
And on Tow-Boot 2023.07 its like this:
```
mdio_register: non unique device name 'ethernet@fe300000'
mdio_register: non unique device name 'ethernet@fe300000'
mdio_register: non unique device name 'ethernet@fe300000'
mdio_register: non unique device name 'ethernet@fe300000'
mdio_register: non unique device name 'ethernet@fe300000'
mdio_register: non unique device name 'ethernet@fe300000'
mdio_register: non unique device name 'ethernet@fe300000'
mdio_register: non unique device name 'ethernet@fe300000'
No ethernet found.
```
Ethernet is also works with Armbian's U-Boot build.
3. I flashed eMMC with regular Debian 12 install and found that for some reason Linux can not mount eMMC during boot if it was started by Tow-Boot or upstream U-Boot 2022.07/2023.07, even with kernel's own devicetree (and there is same issue with microSD as far I can tell). However, somehow there is no such issue with Armbian's U-Boot 2020.04 build (I have old build of Armbian for NanoPC-T4 which I used for this test) - it let Linux 6.4 mount eMMC during boot. 

Any idea how come that microSD and eMMC is broken with upstream U-Boot on NanoPC-T4 (probably for years?) but not on Renegade Elite and other boards? Is there anything I been missing, some build option that will make it magically works? Or maybe some kernel module should be built-in?

shvetsnikita, 2023-08-10:
Answer to [this comment](https://github.com/Tow-Boot/Tow-Boot/issues/258#issuecomment-1671914381): 

> And sometimes they ship vendor forks of U-Boot, rather than mainline U-Boot.

As far I understand in case of RK3399 they ship patched upstream 2022.04 instead of vendor fork.

> Can you also try against mainline U-Boot 2023.07, so it's made more obvious whether or not it's a Tow-Boot issue or not?

Already did, as you can see in issue description. Unfortunately all of these is reproducible with upstream 2023.07.

samueldr, 2023-08-10:
> Unfortunately all of these is reproducible with upstream 2023.07.

Or from Tow-Boot's point of view *fortunately* it's not some odd interaction.

* * *

I haven't looked into it at all, but here's clues you could be on the look-out for while digging around in Armbian changes.

### About USB

I would *guess* that some DT config for clocks is missing, and/or driver for the clocks... -19 is -ENODEV.

### About SD / eMMC

My first guess would be that the kernel drivers work on assumption that *some* state from the previous stage set something up in a particular way. That's generally the reason things get odd even when using the same DT across two U-Boot builds.

### About Ethernet

I don't really know, it would involve looking at why it states `No ethernet found` or `non unique device name`, and how other RK3399 have configured their DT for `fe300000` in U-Boot.

It *could* be as simple as an incomplete DT config for ethernet. It could be anything else though, I don't know the networking subsystems of U-Boot at all.

shvetsnikita, 2023-08-11:
> I haven't looked into it at all, but here's clues you could be on the look-out for while digging around in Armbian changes.

Here is what I found so far: [dwc2 patch](https://github.com/armbian/build/blob/main/patch/u-boot/u-boot-rockchip64-v2022.04/general-dwc-otg-usb-fix.patch), [USB enablement for NanoPC-T4](https://github.com/armbian/build/commit/320bf63430b442700cf1cea7e290b9ef7bdde0e9), [rockchip_dw_mmc patch](https://github.com/armbian/build/blob/main/patch/u-boot/u-boot-rockchip64-v2022.04/sdmmc-force-fifo-mode-in-spl.patch), [clk_rk3399 patch](https://github.com/armbian/build/blob/2813365dd25e3ad110936cbf014b95b38d7090ec/patch/u-boot/u-boot-rockchip64-v2022.04/rk3399-always-init-rkclk.patch).

While USB is detected with these patches (but only on Tow-Boot 2021.10 and 2023.07; if I didn't mistake anywhere, somehow  USB is remain broken on Tow-Boot 2022.10 even with patches) nothing else is changed, so Linux still doesn't detect eMMC when I try to boot it from eMMC. Also this time I tested boot from microSD, and it fail with "SD Boot failed. Press any key to continue..." message, which is a bit strange, because I don't see this issue with upstream U-Boot (with upstream U-Boot kernel is started, but doesn't detect microSD, so it can't mount rootfs).

During USB boot testing I accidentally noticed that it's doesn't work not only with Tow-Boot 2021.10 and 2023.07 (with mentioned patches) but also even with Armbian's U-Boot 2022.04 build. It's either error messages like "Unexpected XHCI event TRB, skipping...", or stacktrace that ends with Resetting CPU, or simply freeze after exiting boot services.

I didn't look into Ethernet support yet.

shvetsnikita, 2023-08-11:
Upon further testing I found that power cut off or board reset button is not sufficient for getting reproducible result during Linux boot, at least with mainline kernel. On same combinations of kernel version and dtb I got full range of different results, from not detected microSD, init freeze during boot, to normally booted OS with working GUI, hardware accelerated graphics, Ethernet connection, detected NVME, etc. (That was with U-Boot 2022.04 from Armbian.) So it's seems like from now on I should disconnected everything from board on every power on attempt to get somehow reliable information.

Unfortunately, in practice it helps me get more reliable boot flow with Armbian's U-Boot build (2020.04, not even 2022.04) but not with Tow-Boot or upstream U-Boot - Linux still doesn't detect neither microSD or eMMC.