Tow-Boot/Tow-Boot #261: [Pinebook Pro] Screen flashes on boot up

blmayer, 2023-08-16:
I flashed tow-boot using shared image setup. When the laptop boots up the screen flashes white before anything is displayed, this is very annoying, the flash is very intense. Also, after some messages the screen flashes again, this time with some weird pattern, mostly green, and then continues booting. Is anyone having the same issue?

samueldr, 2023-08-17:
For the first flash, it may end-up being fixed by this pair of changes:

 - https://github.com/Tow-Boot/U-Boot/commit/2498667a87bec6a1f0440f9b41bb4a9def2d1f4d
 - https://github.com/Tow-Boot/U-Boot/commit/8f64f99e95d29eccb0b5a7335dfe14748de94335

The first ensures we don't keep whatever was in RAM in the framebuffer before the first write to the framebuffer. If whatever was in RAM was "high values" it would show up as white, or whatever.

The second changes how the display is lit-up, by turning it off first early during boot. This only affects a warm (re)boot.

As for the second, I can't say for sure. I think I might know what you're referring to, and if it is, then that would be by using extlinux-compatible boot. Can you share which OS boot method your install uses?

blmayer, 2023-08-17:
Cool, thanks for taking your time. Are these two commits going on the next release, and when it is going to be released? If not can I just pull master?

> Can you share which OS boot method your install uses?
I'm using archlinux arm image, it installs everything by `dd`ing the install image, so I'm not sure. 

My guess is `extlinux` since I have a folder with that name on `/boot`

samueldr, 2023-08-17:
Planned for next release (unless an interim amlogic bugfix gets written and tested with someone with the hardware to validate); unsure when exactly, but I definitely want it sooner than later.

blmayer, 2023-08-21:
Thanks. Since the second flash is related to extlinux, AFAIK, I'm closing this.

closed 2023-08-21 by blmayer