Tow-Boot/Tow-Boot #278: PinePhone: document a "reset" fix for blank screen during installation

smithfred, 2023-10-17:
On a PinePhone (Beta edition) and Tow-Boot 2022.07-006, on booting from a microSD flashed with `mmcboot.installer.img`, the display remained blank, even though boot completed successfully (same serial output as another PinePhone that had display output, and tapping the screen region for the Tow-Boot installer shutdown button logged a shutdown message).

The same thing happened on multiple attempts even after removing/reinserting the battery, but using the PinePhone's reset button fixed the issue and resulted in display output. Presumably there is some issue sometimes with the display/controller not being in the correct state on shutdown, and Tow-Boot not handling this.

It would be useful to add a note about the above workaround to https://tow-boot.org/devices/pine64-pinephoneA64.html.

quinn-dougherty, 2023-10-17:
Is this a pressing the down-volume button situation? I would've liked the docs to have put "`Booting an operating system from an SD card will require holding the volume down button before and during the second vibration. When done successfully, the LED will turn aqua.`" earlier in the page instead of at the bottom, cuz it's such a mandatory thing!

samueldr, 2023-10-17:
It sounds like a different issue; here @smithfred seems to always be able to boot the installer image (which does not require holding a volume key on allwinner systems)... Though the installer system seemingly doesn't setup the display our, or maybe the backlight is stuck off.

In any case, it's odd.