Tow-Boot/Tow-Boot #281: v2023-007 release

samueldr, 2023-10-18:
See:

 - https://github.com/Tow-Boot/Tow-Boot/milestone/8

* * *

We're basically there. `-rc3` is what I expect to be release for all devices, except the Pinephone Pro, which will be the only device with changes in `-rc4` (#280).

Report testing in this issue, if you want to help.

tpwrules, 2023-10-19:
Validated that boot works properly running on an SD card on ODROID-C4.

AndreySV, 2023-12-10:
@samueldr Is there any built, that you'd like to test? Or should I build it myself?

AndreySV, 2023-12-30:
I've tested 2023.07 007-rc3-pre on original PinePhone. I can confirm, that it boots on the device, but I wasn't able to enter tb_menu over serial console by pressing ESCAPE or CTRL-C like it was before.  Serial output hang right before 
```
Core:  75 devices, 21 uclasses, devicetree: separate
WDT:   Not starting watchdog@1c20ca0
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.

``` and then jumps to U-Boot menu.

samueldr, 2024-03-31:
> 1. 2023-007-rc3 detect both board revisions as 1.2. As a result magnetometer isn't working on 1.2b boards.

(Sorry if it's a bit late)

Can you confirm that this is not a *regression* compared to the previous release? If it's not a regression, it'll be handled in the next release cycle.

samueldr, 2024-03-31:
> It may be a good idea to mention the random boot-failures of the PinePhone in the installation-instructions, so people don't assume that Tow-Boot does not work, if it randomly happens then.

@MoonlightWave-12: Can you describe more about those random boot failures? As many details as you can observe. I'm not sure I know about those "usual" random boot failures.

samueldr, 2024-04-02:
> will vibrate 10 times

Interesting... This signals that Tow-Boot is fully running, but found nothing out of the default boot chain to boot (and then shuts down to not drain the battery).

I'll try booting a few times. What distribution are you using? Installed to eMMC or to SD?

samueldr, 2024-04-05:
I've hidden your reply, and opened an issue to try and keep track of this, considering it is not a regression. Thank you for the all the information, it will help ensuring I reproduced the right issue.

AndreySV, 2024-04-09:
> > 1. 2023-007-rc3 detect both board revisions as 1.2. As a result magnetometer isn't working on 1.2b boards.
> 
> (Sorry if it's a bit late)
> 
> @AndreySV: Can you confirm that this is not a _regression_ compared to the previous release? If it's not a regression, it'll be handled in the next release cycle.

This is not a regression. Let's do that in the next cycle.

closed 2024-04-18 by samueldr