Tow-Boot/Tow-Boot #284: Add Support For Pine64 & Radza RK3566 Boards

RFWave, 2023-11-06:
These SBCs are now supported in mainline U-Boot after v2023.10. Are they now able to be added in Tow-Boot?

Quartz64 Model A
Quartz64 Model B
SOQuartz
Radxa CM3

xdevs23, 2025-02-17:
The Radxa Zero 3 is also RK3566. I'm trying to build tow-boot for it. I added the rk3566 hardware definition and the zero 3 board but so far I haven't been able to build it, getting this error message:

```
Built /build/source/build/rk3399/release/bl31/bl31.elf successfully

  OD      /build/source/build/rk3399/release/bl31/bl31.dump
installing
post-installation fixup
shrinking RPATHs of ELF executables and libraries in /nix/store/zjqscymv5cw7f36j39cdss29shsy2d7p-arm-trusted-firmware-rk3399-2.10-aarch64-unknown-linux-gnu
shrinking /nix/store/zjqscymv5cw7f36j39cdss29shsy2d7p-arm-trusted-firmware-rk3399-2.10-aarch64-unknown-linux-gnu/bl31.elf
patchelf: cannot find section '.dynamic'. The input file is most likely statically linked
checking for references to /build/ in /nix/store/zjqscymv5cw7f36j39cdss29shsy2d7p-arm-trusted-firmware-rk3399-2.10-aarch64-unknown-linux-gnu...
patchelf: cannot find section '.dynamic'. The input file is most likely statically linked
patching script interpreter paths in /nix/store/zjqscymv5cw7f36j39cdss29shsy2d7p-arm-trusted-firmware-rk3399-2.10-aarch64-unknown-linux-gnu
building '/nix/store/jmwlvjhb3m9361mr41537b8k76v51ikn-Tow-Boot-radxa-zero-3-rk3566_defconfig-noenv-aarch64-unknown-linux-gnu-2023.07-008-pre.drv'...
unpacking sources
unpacking source archive /nix/store/9147vdwpbg0x1s3wfczsgqlcxmgkg18x-Tow-Boot
source root is Tow-Boot
patching sources
:: Dropping EXTRAVERSION from Makefile
sed: can't read Makefile: No such file or directory
/nix/store/355pb9ck3jdnkwcp4zydrnz4pp7ix72w-stdenv-linux/setup: line 144: pop_var_context: head of shell_variables not a function context
error: builder for '/nix/store/jmwlvjhb3m9361mr41537b8k76v51ikn-Tow-Boot-radxa-zero-3-rk3566_defconfig-noenv-aarch64-unknown-linux-gnu-2023.07-008-pre.drv' failed with exit code 2;
       last 7 log lines:
       > unpacking sources
       > unpacking source archive /nix/store/9147vdwpbg0x1s3wfczsgqlcxmgkg18x-Tow-Boot
       > source root is Tow-Boot
       > patching sources
       > :: Dropping EXTRAVERSION from Makefile
       > sed: can't read Makefile: No such file or directory
       > /nix/store/355pb9ck3jdnkwcp4zydrnz4pp7ix72w-stdenv-linux/setup: line 144: pop_var_context: head of shell_variables not a function context
       For full logs, run 'nix log /nix/store/jmwlvjhb3m9361mr41537b8k76v51ikn-Tow-Boot-radxa-zero-3-rk3566_defconfig-noenv-aarch64-unknown-linux-gnu-2023.07-008-pre.drv'.
error: 1 dependencies of derivation '/nix/store/fv4dgbbrphjq01gv8fmy335m527jzgwm-Tow-Boot.radxa-zero3.2023.07-008-pre.drv' failed to build
```

xdevs23, 2025-02-18:
It turns out I accidentally specified the incorrect directory as src. I need to specify u-boot source but I accidentally spcified Tow-Boot source. Oops.

xdevs23, 2025-02-18:
I can't get it to work...
```
       > substitute(): ERROR: file 'include/tow-boot_env.h' does not exist
       > /nix/store/355pb9ck3jdnkwcp4zydrnz4pp7ix72w-stdenv-linux/setup: line 144: pop_var_context: head of shell_variables not a function context
       For full logs, run 'nix log /nix/store/ljzhk91jffr93fp36cm1sh3d44w97wrm-Tow-Boot-radxa-zero-3-rk3566_defconfig-noenv-aarch64-unknown-linux-gnu-2023.07-007.drv'.
error: 1 dependencies of derivation '/nix/store/iyc7cbn64qrn8s0m8z54d1v9kkm87ang-Tow-Boot.radxa-zero3.2023.07-007.drv' failed to build
```

tow-boot and the u-boot fork in tow-boot org hasn't been updated for a while. I'm going to use regular u-boot for now then. I can try using the buildUBoot thing in NixOS.