Tow-Boot/Tow-Boot #285: Manual installation of tow-boot via OTG port on Radxa Rock Pi 4

resdigita, 2023-11-07:
I'm looking for a way to install tow-boot via OTG port (without booting via mSD card into thegraphical installer)

On a Radxa Rock Pi 4plus the graphical installer does not react on keyboard input. So an installation is not possible. Furthermore, with a NVMe board installed, the connection cable blocks access to the mSD connector., So you can't boot from nSD anymore. 

So the only way to install or to update is using the OTG USB port. And the question is, how or where to get the spi software to transfer and which Offset to use. 

Any info appreciated.

wdthompson, 2023-11-09:
I had to do a manual install, the uboot version on emmc "lost" the emmc after kernel, dtb load so would ONLY boot SD
(using emmc uboot and then unable to erase it)  lsblk showed no emmc.. 
install mtd-utils, nandtest is a good but slow way to erase spi,,, -k to keep data
Some advice is to write to /dev/mtd0, this likely is wrong, if lsblk shows mtdblock0 this is correct
something like ... dd if=Tow-Boot.spi.bin of=/dev/mtdblock0 status=progress ; sync
You can use balena-etcher, in 2nd screen check show system drives, it writes (for me) to mtdblock0
Tow-Boot.spi.bin is in the binary subdirectory, about 1.5MB
If you have serial spi can be flashed from uboot serial terminal, the commands have strange syntax
which I barely know

wdthompson, 2023-11-09:
--more--
If you want to be sure,, do a nandump of the spi of the EXACT length of Tow-Boot.spi.bin to a tmp file, 
compare the md5sum of the 2 files