Tow-Boot/Tow-Boot #286: Doesn't boot - PinePhone

realroot2185, 2023-11-10:
I have pmOS in the external SD which for this device takes over tow-boot in SPI.

I zeroed it as the wiki states:
`dd if=/dev/zero of=/dev/mmcblk2 bs=8k seek=1 count=4`

After when booting and pressing volume down, that triggers boot from external SD, I saw aqua led, white led, flashing red led.
(Aqua led is when it starts booting from external SD).

Do I need to do something? Changing `boot.scr` etc.

realroot2185, 2023-11-11:
Same problem with Artix and [Arch](https://github.com/dreemurrs-embedded/Pine64-Arch/pull/399).

ArenM, 2023-11-15:
As far as I'm concerned the issue with arch is a buggy boot script, not tow-boot.

The problem in that case stems from `mmc_bootdev` being set to the emmc instead of the sd card, which seems to be the intended behavior of mmc_bootdev. The solution in that case is to replace mmc_bootdev with devnum.

realroot2185, 2023-11-15:
Do I simply replace all instances of `mmc_bootdev` with `devnum`? I am using Arch boot.txt.

realroot2185, 2023-12-27:
It works for Artix on the external SD too.
I wonder why Arch package doesn't change that one word.

closed 2023-12-27 by realroot2185