Tow-Boot/Tow-Boot #287: Rockchip TPL support

Haxk20, 2023-11-11:
Hello,

Currently some devices need an rockchip provided binary which initializes the dram instead of the usually shipped uboot TPL.
U-boot added support for this few months ago (https://github.com/u-boot/u-boot/commit/05b978be5f5c5494044bd749f9b6b38f2bb5e0cc) and this is used on PineBook-Pro to fix suspend.

Would it be possible to port this function (Probably make it optional) so that we can have actual suspend supported on Tow-Boot ?

I looked into it myself but its the first time i have seen anyone use NIX for such project and im not familiar with it. Thus the issue :)

samueldr, 2023-11-11:
Yes it is entirely possible. Even without outright U-Boot support.

I have three WIP branches that do just that.

Though one thing that is a bit "spicy" here is that by default those use the `1_500_000` (1.5 million) baud rate, instead of the harmonized `115_200`.

... but that's not that much of an issue since I explored *enough* of the structure of their ram init, and the structure used to configure serial are trivial to find, and easy to patch out.

* * *

As far as Pinebook Pro support goes, it would need to be properly investigated to see if it fixes suspend. (Also first verifying suspend is broken and under which conditions...)

* * *

While using the "open source" RAM init is preferred, I personally believe that using the proprietary bit is fine when it actively makes things better. After all, the "open source" RAM init is still just a bunch of data sent to the RAM controller.

In an ideal world, we'd have people that know a bit better about how the Rockchip TPL works look into automatically dumping the RAM init data from them, such that it becomes less of an issue overall.

Haxk20, 2023-11-11:
> Yes it is entirely possible. Even without outright U-Boot support.
> 

That is wonderful 
> I have three WIP branches that do just that.
> 
> Though one thing that is a bit "spicy" here is that by default those use the `1_500_000` (1.5 million) baud rate, instead of the harmonized `115_200`.
Yea RK being RK...
> ... but that's not that much of an issue since I explored *enough* of the structure of their ram init, and the structure used to configure serial are trivial to find, and easy to patch out.
>
> * * * [...]

Basically always xD. I got my PBP and suspend worked flawlessly. Installed tow-boot on spi and its gone again :)

I then shorted out the SPI so it boots from emmc again and flashed mainline uboot (without the binary) and again suspend is borked. I then reflashed the one with the binary and we were back in business suspending properly :)

> * * *
>
> While using the "open source" RAM init is preferred, I personally believe that using the proprietary bit is fine when it actively makes things better. After all, the "open source" RAM init is [...]

Well here is where it becomes a bit more of a problem situation xD

This is the command to produce suspend working u-boot:
> `make O=out CROSS_COMPILE=aarch64-linux-gnu- BL31=../rkbin/bin/rk33/rk3399_bl31_v1.36.elf ROCKCHIP_TPL=../rkbin/bin/rk33/rk3399_ddr_933MHz_v1.30.bin`

As you can see it has to also use bl31 binary from rockchip.

Thats why I said that when this feature is implemented it should be made optional via some environment flag while building or etc. i think providing prebuilt binary with  the binary blobs from rk is fine given that they do fix suspend and users would prefer working laptop over having to compile tow-boot to enable it.

samueldr, 2023-11-12:
> As you can see it has to also use bl31 binary from rockchip.

Using the proprietary TPL does not *require* using the proprietary TF-A/ATF/BL31.

And this is somewhat related to what I meant by investigating *what* solves the suspend issues exactly. There's some gaps in my knowledge to be 100% sure of this, but I believe that TF-A is what matters in your situation. So it's not only about the TPL, but also using their TF-A build.

In this case, it's just like how it is here:

 - https://github.com/Tow-Boot/Tow-Boot/blob/3436997d2904225d75acf2e9c76c58f78ac4bf57/modules/hardware/rockchip/default.nix#L70

But using the proprietary build instead of the mainline one.

Selecting a different TPL is/will be basically the same once we use the new~ish bits in U-Boot that assemble the image.

Haxk20, 2023-11-12:
> > As you can see it has to also use bl31 binary from rockchip.
> 
> Using the proprietary TPL does not *require* using the proprietary TF-A/ATF/BL31.

Thats not at all what i said. I said that to have suspend the bl31 from them is needed.
> And this is somewhat related to what I meant by investigating *what* solves the suspend issues exactly. There's some gaps in my knowledge to be 100% sure of this, but I believe that TF-A is what [...]

Seems simple enough to modify locally :)
> 
> Selecting a different TPL is/will be basically the same once we use the new~ish bits in U-Boot that assemble the image.
Any idea how fast we could start using the new uboot bits for the image ?

samueldr, 2023-11-13:
> Thats not at all what i said. I said that to have suspend the bl31 from them is needed.

Right, that was a poor wording on my end. I meant that TPL vs. BL31 are discrete pieces. It might be that the proprietary BL31 also depends on the proprietary TPL, but if it doesn't, switching BL31 can already be done without adding the necessary plumbing for the TPL.

wdthompson, 2023-11-15:
Yes ATF or TF-A does indeed have something to do with it, at least for STR, 14d sleep
Are we talking 1 day sleep or sleep to ram? (14d) Most people seem NOT to differentiate 
With tow-boot v2,,  STR worked, oddly  ~7-8d,,, v3 returned a blank screen, so one has NO idea
I suppose I could have tried ssh, didn't think of it, the back has NEVER been off, so no serial
pbp (2019-2020)

sado1, 2024-01-07:
FYI: if I am right (as I have zero knowledge in this area), using proprietary TPL and TF-A would also let PinePhone Pro have much lower idle power consumption, due to DRAM downclocking support.

Details in https://xnux.eu/log/083.html (and partially in https://xnux.eu/log/091.html )