Tow-Boot/Tow-Boot #293: Tow-Boot doesn't differentiate 1.2 and 1.2b board revisions for Original PinePhone

AndreySV, 2024-01-03:
In newer 1.2 PinePhone board revisions LIS3MDL magnetometer was replaced by
AF8133J. They use the same PB1 pin in different modes.

LIS3MDL uses it as an gpio input to handle interrupt.
AF8133J uses it as an gpio output as a reset signal.

It wasn't possible at runtime to enable both device tree
nodes and detect supported sensor at probe time. As a result new board
revision was introduced with AF8133J definition (1.2b).

2023-007-rc3 detect both board revisions as 1.2. As a result magnetometer isn't working on 1.2b boards.

AndreySV, 2024-01-03:
Patch introduced new board revision:
https://salsa.debian.org/Mobian-team/devices/kernels/sunxi64-linux/-/blob/mobian-6.6/debian/patches/pinephone/0149-arm64-dts-allwinner-sun50i-a64-pinephone-Add-support.patch?ref_type=heads

Issue about non-working magnetometer on PMOS:
https://gitlab.com/postmarketOS/pmaports/-/issues/1945

samueldr, 2024-04-18:
Hi, sorry it took a while until I got back to this issue.

Adding a quick note to self: (new) related PR: https://github.com/Tow-Boot/U-Boot/pull/2