Tow-Boot/Tow-Boot #301: PineBook Pro - Booting with Grub/EFI from NVME is unreliable and sometimes the kernel boot gets frozen in the initramdisk

jcapik, 2024-02-04:
I'm unsure whether this issue is related to Tow Boot or something else (grub or kernel). However, It's good to track it at least.
After flashing the Tow Boot to SPI and booting from SATA drive connected via USB->SATA adapter, everything worked well. After moving everything from the SATA drive to internal NVME, the Grub always loads without problem, but the Linux boot sometimes fails and I end up in the initramfs rescue environment. That happens for 2 different distributions (Armbian and Manjaro) having two different kernels.
Any chance to debug / troubleshoot this?

DrYak, 2024-04-19:
Note that it could also be a [hardware limitation of the PineBook Pro](https://wiki.pine64.org/wiki/Pinebook_Pro_Hardware_Accessory_Compatibility#NVMe_SSD_drives):
 - the PBP cannot provide more than 2.5W on the 3v power that it provides to NVMe
 - the 3v gets noisy if battery power drops too much
 
 In addition to that, on the NVMe side:
 - Not all NVMe can run nicely on power settings that are lower than 2.5W (some run extremely slowly and require a power setting higher than 2.5W for good performance)
 - Not all NVMe can _**save**_ the power saving setting, meaning that on each reboot one need to set the power setting.

U-Boot itself doesn't manage power settings. And thus what happens is that U-Boot does successfully initialize your NVMe, but your NVMe runs at a too high power consumption (e.g. 3.0W) and the whole system is unstable. It kind of suffice for U-Boot to load GRUB, and then that latter loads the kernel and the initramfs, but during Linux's boot with everything going on, the system borwns-out and the NVMe fails. Or, the NVMe at boot uses a power setting that is barely passing (e.g. 2.60W), but the battery is low (<3.9V) the 3V power is too noisy and that is the thing that kills your NVMe.

So given the above, check your NVMe drive and you might need to:
 - Always have enough percents of battery left when booting, or plug it into a powerbank during boot time.
 - Buy another NVMe which is more power frugal (e.g.: I my self have bought a Intel 660p M.2 512GB which as PS2 consumption off 2.0W)
 - If the drive supports it, set the power setting (e.g. to PS1 or PS2) and have it saved so on subsequent power-on, the NVMe directly starts at a power level which is fine for the PineBook Pro.
 - If the drive DOES not support saving the power state, store GRUB, the Kernel and the initrd (basically the whole `/boot` partition) on the eMMC drive and have the initrd change the power settings on each boot. So by the time the root and/or home are mounted, the drive is in a power settings that is compatible (e.g.: I use that approach as my Intel 660p cannot save power settings)

reukiodo, 2024-11-15:
Is it possible for Tow-Boot to initialize the drive in a low power state by default?