Tow-Boot/Tow-Boot #303: [Broad wanted] Orangepi4lts(RK3399)

calvinweb, 2024-03-15:
is it possible to run tow-boot on it?

CRTified, 2024-03-15:
I don't see a reason why it shouldn't work. It's an RK3399-T, I have a board with the same SoC at hand and want to get Tow-Boot on it as well.

Only disadvantage is that there is apparently no SPI flash on it, at least I was not able to spot one on [the product page](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/orange-pi-4-LTS.html). That would mean tow-boot has to run in the shared storage approach, being flashed on the µSD/eMMC along the OS image.

calvinweb, 2024-03-16:
> I don't see a reason why it shouldn't work. It's an RK3399-T, I have a board with the same SoC at hand and want to get Tow-Boot on it as well.
>
> Only disadvantage is that there is apparently no SPI [...]

Cool, BTW Is it possible to flash it into emmc via rkdevtool?

CRTified, 2024-03-17:
I honestly never used rkdevtool or eMMC as storage before, so I can't really tell. But if you can use rkdevtool to flash OS images (i.e., doing the same thing as `dd`ing some stuff to the eMMC), I don't see a reason why it won't work, as the shared storage approach does exactly that - storing tow-boot on the same storage as the OS image.

calvinweb, 2024-04-03:
> I don't see a reason why it shouldn't work. It's an RK3399-T, I have a board with the same SoC at hand and want to get Tow-Boot on it as well.
>
> Only disadvantage is that there is apparently no SPI [...]

Hello, last question, have you tried to use USB disk boot? I just got the following logs when booting linux:
```
EFI stub: Booting Linux Kernel...
EFI stub: EFI_RNG_PROTOCOL unavailable
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...
```