Tow-Boot/Tow-Boot #304: Mud-Boot – for platforms that cannot be supported well

mskiptr, 2024-03-22:
## Intro
Today I was watching the [FOSDEM talk](https://fosdem.org/2024/schedule/event/fosdem-2024-1716-u-boot-for-modern-qualcomm-phones/) by @calebccff about U-Boot on ABL devices. Even though the bootloader situation there is grim and scary, the prospect of layering our own EFI runtime is a real ray of hope. Once materialized, that should be a massive improvement to packaging and boot infrastructure on upstream-first OSes like pmOS or Mobian.

## What this is about
- a lot of devices have their own complicated boot process we cannot get rid of or replace
- if we could unify all of them (from the OS perspective) by flashing a device-specific payload, that would be great
- "distribution of U-Boot" – essentially what Tow-Boot is
- Tow-Boot is high-quality; this is not
- what if that perfectionism is why Tow-Boot supports only a narrow selection of boards?
  - (looks like that from the outside at least)
- keeping Tow-Boot consistent and polished is a great thing, but we could also use something more messy as a staging area

_"Why waste time say lot word when few word do trick?"_ – after listing out all the points I want to write about, I realized that this list perfectly sums up what I wanted to say. So why would I make it more complicated by now re-writing it all into proper paragraphs? Also, making perfect the enemy of good would go directly against a major point of this post. If something is unclear, please ask for a better explanation.

## Side-ideas
We could also have a flasher program that figures out what state the device is in, then flashes what needs to be flashed, re-partitions the device if that's a good idea, maybe knows some exploits to liberate locked-down devices and overall guides the user through the process.

A major point here is that the OS should not have to know about all these quirks. It should just put its EFISTUB kernels or a boot manager EFI app in `\EFI\$distro\` of the ESP, ideally register it with the EFI runtime somehow and be done with it. No more bootloader flashing from the package manager. The OS doesn't "own" the global state and should not meddle with it if not asked to. Maybe teaching fwupd about these bits of 'firmware' would be a good idea. Or maybe this is too messy and it's better to leave all this complexity to that flasher from above (and do self-flashing?).

## The name
So, about the name… I was trying to come up with one that would be associated with messiness and yet would still be somewhat similar to Tow-Boot. The first short word that came to my mind was "mud". I googled "Mud-Boot" to see if there's anything already called like that and to my surprise, instead of seeing a bunch of info about boats, all I got was pictures of muddy boots.

Perfect! We can think of it as a trusty pair of boots that protect us from all the mud we're standing on.

As you can see I had way too much fun with this name. And that's frankly _the_ reason I wrote all of this (instead of shelving the idea on my long list of stuff to do in the future™).

ArenM, 2024-03-23:
I don't think any of what your describing is outside the scope of tow-boot, at least from my point of view as an outsider. Tow-boot is supposed to be the thing that handles the messy boot process so a distro doesn't have to care what hardware it's on. It's configured to detect and just boot any distro puts an efi image or config file in the right spot, and it even has some integration with fwupd already. I don't even think unfinished device support is that big a deal, as long upstream u-boot can boot the device.

With all that said, you have me convinced that mud-boot is a fun name, and could be great for something like a testing channel / nightly builds.

mskiptr, 2024-03-23:
The reason I assumed it has to be a separate thing is that right now Tow-Boot only supports SBCs with a relatively clean and straightforward bootloader situation (+ the PinePhones and PineBooks, but they are essentially the same in this regard).
On these boards, we are in control of the whole boot chain – everything after the bootrom is replaceable – and so Tow-Boot has all the opportunities to shine.

Android devices already come with their own bootloader. One that is usually buggy, complicated and not flexible enough. But because in general we cannot get rid of, our only option is layering a second-stage bootloader on top. In the talk that I mentioned at the beginning @calebccff suggested using U-boot with its EFI implementation for this purpose. And since this is very similar to the mission of Tow-Boot, I think starting here would be the best course of action.

Keeping all the boot firmware in a dedicated place and leaving the entire main storage for the OSes is an important UX consideration. You can erase it all, yet the device will still turn on and let you re-install the OS without specialized tools. You can't brick your device by e.g. re-partitioning it.

Since we cannot uphold the same guarantees on these 'messier' platforms, I believe it would be better to keep the 'branding' separate. It would clearly set the correct expectations in both cases. Still, it could very well be developed in the same code base and as part of the Tow-Boot project. When the device is fragile like this, it might require a more careful and complicated approach, but even then a lot of the code could likely be shared.

samueldr, 2024-03-25:
I was considering implementing the messier U-Boot for Qualcomm-based phones into Tow-Boot proper, once I'd had the time and energy to dedicate into understanding the pitfalls.

Even if it needs adding a new "class" of devices.

mskiptr, 2024-03-25:
Great to hear from you @samueldr

I hope to help with this effort (after I learn more how Tow-Boot is built – I'll be trying to bring up Libre Computer's RK3328 board).

calebccff, 2024-03-28:
Hello!

I think that we can do better here. I'm not as affluent at the upstreaming process as I wish I was, but regardless I think that we can do almost all that we need/want to in upstream U-Boot. For the Qualcomm devices I have so far implemented a few things that I think make great strides for Tow-Boot, of the top of my head:

* The ability to map buttons to run arbitrary commands ([this feature is upstream!](https://source.denx.de/u-boot/u-boot/-/blob/master/common/button_cmd.c)).
* Automatically mapping volume/power buttons to up/down/enter ([not upstream but I plan to submit it](https://git.codelinaro.org/linaro/qcomlt/u-boot/-/commit/97784f31502f36fbf3222a8cf4aa337b58510b5c)).

I also think that having a "persistent" boot menu (so you never drop to a prompt) would be a relatively easy feature to add, I have some PoCs for this that really only require changing like 2 lines of code to re-enter the boot menu automatically.

The impression I get from a lot of the existing U-Boot board support is that a lot of it is a holdover from when U-Boot *require* writing board specific code. But this is just not the case anymore -- as of a few days ago it's now possible to have a single U-Boot image (installed via fastboot) [which can be flashed to half a dozen different Qualcomm devices](https://git.codelinaro.org/linaro/qcomlt/u-boot/-/commit/785ce07a45e7945781962cafa23993f4ded58a6f). It's of course a slow process, and avoiding breakage it hard, but I think we'll see Rockchip and sunxi support move in this direction too.

# Some overall thoughts on this thread

## Automatic U-Boot updates

I've spoken to Richard (of fwupd/LVFS) and some other folks both in industry and across the community. And the way things seem to be going, I see absolutely no reason why we can't eventually have 100% upstream U-Boot supported on devices like the OnePlus 6 (as well as msm8916 phones and plenty others) with updates provided automatically and in a distro-agnostic way.

Most recently I've been looking into generating firmware UUIDs dynamically based on the device model (so we don't need to maintain a hardcoded list in U-Boot itself) ([see here](https://git.codelinaro.org/linaro/qcomlt/u-boot/-/commit/0ab749546c312fbf2dc71e3c3434201b383378d3)). We would provide updates on a community fwupd channel and with this we'd be able to trivially support a new device just by adding it's DTB to the U-Boot image (or building a device specific one if need be) and adding it to LVFS.

## Installing U-Boot

First-time installation is a tricky problem, we're obviously limited by whatever firmware is initially provided by the vendor. And it is _far_ too easy to get stuck in the weeds trying to build something agnostic here. For qualcomm devices we can at least have something like <https://flash.postmarketos.org/>, as U-Boot is just flashed via fastboot.

I don't think it's reasonable to try and abstract this away from users. Beyond a certain point we're doing ourselves a disservice when we don't just say "hey, the manufacturer of this device sucks and they make it difficult to install your own software. We tried our best but you should vote with your wallet next time and/or complain to them".

## Branding

To be honest, I would rather Tow-Boot didn't attempt to support Qualcomm platforms. I know upstreaming is slow going (I have ~120 patches in my branch...) but the situation is really good here, upstream DTBs are fully supported on Qualcomm, and all of the upstream DTS files are now synced to U-Boot. There is tentative work on newer SoCs going on, and older ones should be similarly trivial to support.

You'll find several [opinionated configs and default environments](https://git.codelinaro.org/linaro/qcomlt/u-boot/-/tree/caleb/rbx-integration/board/qualcomm) in `board/qualcomm` in my integration branch which I fully intend to upstream as well.

Ideally, we can provide a consistent UX across Qualcomm devices. If you want to support a new device/platform, you can send clock/pinctrl drivers directly upstream, I will apply them upstream, and they will "just work" with my feature branch.

## Get involved

Currently the main things that are lacking are documentation (as always :/ ) and also stuff like build system integration for generating Android boot images -- this would simplify the build process a lot as it's kinda convoluted right now.

We just merged support for a ["generic SDM845"](https://gitlab.com/postmarketOS/pmaports/-/merge_requests/4599#note_1830235961) devices in postmarketOS, so it's now easier than ever to test out U-Boot based pmOS for yourself.

I'm really excited for what the future holds here, and especially getting LVFS up and running would be really huge!