Tow-Boot/Tow-Boot #315: ROC-RK3399-PC Unable to Install

Zaf9670, 2024-05-20:
I have the ROC-RK3399-PC with Mezzanine. I'm trying to go from the official Libre [u-boot](https://boot.libre.computer/release/roc-rk3399-pc/)  2023.07+ (Oct 13 2023) to the most recent tow-boot but it always seems to fail.

When I try to do it without any SPI/eMMC boot it does not give any feedback over the serial connection or seem to work.

Here is the output from the SD card flashed with spi-installer.img which seems to point to "No menu system available"

```
U-Boot 2023.07+ (Oct 13 2023 - 16:24:18 -0400) Libre Computer ROC-RK3399-PC

Model: Libre Computer ROC-RK3399-PC
Net:   eth0: ethernet@fe300000
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3a0000: USB OHCI 1.0
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe3e0000: USB OHCI 1.0
Bus usb@fe900000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3a0000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 2 USB Device(s) found
scanning bus usb@fe3e0000 for devices... 1 USB Device(s) found
scanning bus usb@fe900000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ----------------
Scanning global bootmeth 'efi_mgr':
Scanning bootdev 'mmc@fe330000.bootdev':
Card did not respond to voltage select! : -110
Scanning bootdev 'mmc@fe320000.bootdev':
** File not found /boot.bmp **
  0  script       ready   mmc          2  mmc@fe320000.bootdev.part /boot.scr
** Booting bootflow 'mmc@fe320000.bootdev.part_2' with script

Tow-Boot SPI installer script

Unknown command 'tb_menu' - try 'help'
Unknown command 'bootmenu' - try 'help'

****************************
* No menu system available *
****************************

Use `run spi_erase` to erase the SPI.

Use `run spi_flash` to erase and flash to SPI.

Use `reset` to reset the system.

Boot failed (err=-14)
rockchip_pcie pcie@f8000000: PCIe link training gen1 timeout!
rockchip_pcie pcie@f8000000: failed to find ep-gpios property
scanning bus for devices...
No more bootdevs
---  -----------  ------  --------  ----  ------------------------  ----------------
(1 bootflow, 1 valid)
=>
```

Zaf9670, 2024-05-23:
As a side note. I do currently have a PCIe card in it. I have not tried removing it for the install. It is an Intel WiFi adapter I wanted to try to add to each device. So that might be the USB device initiations for the Bluetooth module of the PCIe card.

caniko, 2025-03-21:
Curious if this is fixed I was going to install tow-boot.

caniko, 2025-03-25:
Removing the eMMC module fixed it for me