Tow-Boot/Tow-Boot #319: [pine64-pinebookpro] can not boot from usd when towboot 2023.07-007 on SPI

reukiodo, 2024-07-17:
I have several uSD cards (different vendors, different sizes), and Manjaro on eMMC.

When towboot is NOT on the SPI, all uSD cards boot correctly and Manjaro can not.
When towboot IS on the SPI, Manjaro can boot successfully from eMMC and NONE of the uSD cards can. The one uSD exception is the towboot installer uSD card, which can still boot both to the bootmenu and to its flash/erase menu.

When 'Boot from SD' is selected in towboot, this is a sample of the output:
```
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scf
3920 bytes read in 6 ms (637.7 KiB/s)
## Executing script at 00500000
Boot script loaded from mmc 1:1
219 bytes read in 4 ms (52.7 KiB/s)
21077419 bytes read in 2214 ms (9.1 MiB/s)
30028288 bytes read in 3152 ms (9.1 MiB/s)
87012 bytes read in 19 ms (4.4 MiB/s)
Working FDT set to 1f00000
2825 bytes read in 13 ms (211.9 KiB/s)
Applying kernel provided DT fixup script (rockship-fixup.scr)
## Executing script at 09000000
Trying kaslrseed command... Info: Unknown command can be safely ignored since kaslrseed does not apply to all boards.
Unknown command 'kaslrseed' - try 'help'
Moving Image from 0x2080000 to 0x2200000, end=3f40000
## Loading init Ramdisk from Legacy Image at 08000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    21077355 Bytes = 20.1 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... Bad Data CRC
Ramdisk image is corrupt or invalid
SCRIPT FAILED: continuing...
BootOrder not defined
EFI boot manager: Cannot load any image
SD Boot failed.
Press any key to continue...
```

reukiodo, 2024-07-17:
I tried the trick from https://github.com/Tow-Boot/Tow-Boot/issues/196 thinking that maybe it doesn't recognize the SD card correctly but that didn't help.

reukiodo, 2024-07-17:
I found a difference in that all uSD cards not bootable from towboot use MBR where Manjaro and towboot-spi-installer both use GPT

I don't know if this is the issue but I'd like to test more - are there any other distros witch use GPT?

reukiodo, 2024-07-17:
Trying an older [2022.07-006](https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2022.07-006) towboot version lets the uSD card boot, at least no boot error, but now without any display. It seems even the backlight turns off after a little while, though the power led is still green. It doesn't seem to be the same as https://github.com/Tow-Boot/Tow-Boot/issues/139 though maybe similar to https://github.com/Tow-Boot/Tow-Boot/issues/225 ?

I'm not testing from the SPI, but rather booting the 006 from uSD and interrupting to the bootmenu, then swapping uSD cards out to another (like Armbian) and then selecting `Boot from SD`

reukiodo, 2024-07-17:
Trying [2021.10-005](https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2021.10-005) has the exact problem mentioned in #139

dleslie, 2024-07-30:
Just ran into this. Can't seem to get any images to boot due to the CRC check failing.

dleslie, 2024-07-30:
https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2022.07-006 boots the image but turns off the display when it does so. Hmm.

reukiodo, 2024-11-18:
I also got the debian image from https://d-i.debian.org/daily-images/arm64/daily/netboot/SD-card-images/ (firmware.pinebook-pro-rk3399.img.gz + partition.img.gz) to boot from TowBoot, even though it can't install due to other issues.