Tow-Boot/Tow-Boot #330: Documentation on boot order for the PineBookPro

WhyNotHugo, 2024-08-28:
I'm trying to understand which boot order would be used if I installed Tow-Boot on a PineBookPro, and this seems under-documented.

The [device page](https://tow-boot.org/devices/pine64-pinebookPro.html) indicates the boot order for the SoC itself:

> The SoC startup order for Rockchip systems will prefer SPI, then eMMC, followed by SD last.

And the [home page](https://tow-boot.org/) indicates that if tow-boot were installed on shared storage, it would prioritise booting from that storage:

> Boot order is unsurprising: on shared storage, the storage from which the currently running Tow-Boot is running is prioritized by default.

This tells me that if I install Tow-Boot into the eMMC, it would prioritise booting from the eMMC.

If Tow-Boot is installed into the SPI, will it prioritise booting from the eMMC or the SD card?

WhyNotHugo, 2024-11-11:
I'm happy to contribute the documentation for this, but I simply don't know the answer to my underlying question:

> If Tow-Boot is installed into the SPI of a PineBookPro, will it prioritise booting from the eMMC or the SD card?

samueldr, 2024-11-11:
Hi, sorry it took a while until I replied.

Hmmm...

So, let's start with definitions because this is a commonly confusing part.

Two terms:

 - SoC startup order
 - [operating system] boot order

The SoC startup order is the order ***the SoC*** (the RK3399 here) looks for the platform firmware (e.g. Tow-Boot, U-Boot). This is an immutable fact of the hardware. The SoC will start the platform firmware from the first location that the SoC finds a valid one. It differs from SoC to SoC.

The [operating system] boot order is managed by ***the platform firmware*** (i.e. Tow-Boot) is programmed to look at. This is an order that can be changed, and be made what is wanted.

* * *

Both orders are independently defined. So the documentation is *accurate* in describing it that way. Though it may need some more precision to help understand the nuance.

So, to answer the unasked question “what [...] order does the PBP use?”:

 - The SoC checks and starts the Platform Firmware in order: SPI, eMMC, SD
 - Tow-Boot checks for an operating system in order: eMMC, SD, USB

* * *

Now, what's up with:

> Boot order is unsurprising: on shared storage, the storage from which the currently running Tow-Boot is running is prioritized by default.

To be fair... I think that line is [now] wrong!

It was [authored early on in the project](https://github.com/Tow-Boot/Tow-Boot/blame/c9636c7cc71c9f7203a6996611f95749bfee9126/doc/index.md#L70-L71).

That line should instead be something like

> Operating system boot order is unsurprising: Tow-Boot, by default, boots the operating from the “most internal” to the “most external” storage. It will prefer storage that is least easily removable (e.g. NVMe, eMMC) first, then more removable storage last (e.g. SD, USB).

* * *

Does that help?

reukiodo, 2024-11-15:
How much would the boot process be slowed if the preferred order were reversed? (e.g. USB first, uSD, NVMe, AHCI, eMMC, SPI last)

samueldr, 2024-11-15:
> How much would the boot process be slowed if the preferred order were reversed? (e.g. USB first, uSD, NVMe, AHCI, eMMC, SPI last)

Unsure, but the default order was chosen to be more "robust" and "least surprising".

It is surprising that connecting a USB flash drive to a system and (re)booting boots into it by default instead of the system installed on the device.

Now, as a user, it might be desired to change the default. It's *planned*, but blocked (other than by time) on a few additional things, not everything tracked here. Though some notes:

 - #79 
 - #95

reukiodo, 2024-11-15:
I'm not sure how often people normally have bootable media plugged in, as unbootable media should just cause the loader to skip it.

I date myself to say that it was somewhat more common back in the floppy and optical drive days when external media was the preferred boot device and it made recovery of a dead OS much simpler.