Tow-Boot/Tow-Boot #333: Using RE to boot Tow-boot installer from SD because prw+vol.dn failed.  This is the serial output.

ChodryShan, 2024-09-21:
Using RE to boot Tow-boot installer from SD because prw+vol.dn failed.  This is the serial output.

-------

Tow-Boot TPL 2022.07
Channel 0: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 400000000 mhz 0, 1
lpddr4_set_rate: change freq to 800000000 mhz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

Tow-Boot SPL 2022.07
jedec_spi_nor flash@0: unrecognized JEDEC id bytes: ff, ff, ff
jedec_spi_nor flash@0: unrecognized JEDEC id bytes: ff, ff, ff
Trying to boot from MMC2
NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980

Tow-Boot 2022.07 006 [variant: boot-installer]

SoC: Rockchip rk3399
Reset cause: POR
Model: Pine64 PinePhonePro
DRAM:  3.9 GiB
Initializing Pinephone Pro charger
PMIC:  RK8180 (on=0x80, off=0x08)
Core:  324 devices, 34 uclasses, devicetree: sudo ddd
MMC:   mmc@fe310000: 3, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Model: Pine64 PinePhonePro
Net:   No ethernet found.
starting USB...
Bus usb@fe380000: ehci_generic usb@fe380000: Failed to get clocks (ret=-19)
Port not available.
Bus usb@fe3a0000: USB OHCI 1.0
Bus usb@fe3c0000: ehci_generic usb@fe3c0000: Failed to get clocks (ret=-19)
Port not available.
Bus usb@fe3e0000: USB OHCI 1.0
scanning bus usb@fe3a0000 for devices... 1 USB Device(s) found
scanning bus usb@fe3e0000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
[0mPlease press [[1mESCAPE[0m] or [[1mCTRL+C[0m] to enter the boot menu.
off
off
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:2...
Found U-Boot script /boot.scr
512 bytes read in 3 ms (166 KiB/s)
## Executing script at 00500000

::
:: celun FIT image boot script 
::

devtype = mmc
devnum = 1
bootpart = 2
 :: Auto-booting FIT image -> Reading file9340416 bytes read in 981 ms (9.1 MiB/s)
 -> Attempting boot...## Executing script at 00600000
sha1+ 
 :: Auto-booting FIT image...if=IMAGE.img. of=

[... 421 lines cut ...]
xkbcommon: ERROR: failed to add default include path /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-xkeyboard-config-aarch64-unknown-linux-gnu-2.33/etc/X11/xkb
[ERROR] (xkbcommon_init): error: Could not create xkb_context.
libinput error: /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput: failed to find data files
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. See https://wayland.freedesktop.org/libinput/doc/1.16.4/device-quirks.html for details.
xkbcommon: ERROR: failed to add default include path /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-xkeyboard-config-aarch64-unknown-linux-gnu-2.33/etc/X11/xkb
[ERROR] (xkbcommon_init): error: Could not create xkb_context.
libinput error: /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput: failed to find data files
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. See https://wayland.freedesktop.org/libinput/doc/1.16.4/device-quirks.html for details.
xkbcommon: ERROR: failed to add default include path /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-xkeyboard-config-aarch64-unknown-linux-gnu-2.33/etc/X11/xkb
[ERROR] (xkbcommon_init): error: Could not create xkb_context.
[    5.514194] cdn-dp fec00000.dp: Direct firmware load for rockchip/dptx.bin failed with error -2
[    9.546266] cdn-dp fec00000.dp: Direct firmware load for rockchip/dptx.bin failed with error -2
[   17.610224] cdn-dp fec00000.dp: Direct firmware load for rockchip/dptx.bin failed with error -2
[   33.994230] cdn-dp fec00000.dp: Direct firmware load for rockchip/dptx.bin failed with error -2
[   66.250019] cdn-dp fec00000.dp: [drm:cdn_dp_pd_event_work] *ERROR* Timed out trying to load firmware

The system is going down NOW!

Sent SIGKILL to all processes

Requesting system poweroff
[  115.726693] reboot: Power down
\00

_Originally posted by @DiagonalArg in https://github.com/Tow-Boot/Tow-Boot/issues/259#issuecomment-1661889561_

ChodryShan, 2024-09-21:
Ok

DiagonalArg, 2024-09-21:
@ChodryShan 

> Hmm there is a problem I solve it very easily

It's been a year since I've looked at this.  If you have a solution, I'll go back to it, thanks.

ChodryShan, 2024-09-21:
Hmm good man