Tow-Boot/Tow-Boot #339: `Permission denied` error when building with Docker shim on MacOS

glukki, 2024-11-03:
On MacOS if you try to build with Docker shim the `release-2023.07-007` tag, you get the following error:
```
> ./support/docker/build.sh -A uBoot-sandbox
<unimportant output>

(Unwrapping store paths...)
result -> /nix/store/h0pa6p29lnnvfx4m4v0713syzfg1jfyy-Tow-Boot-sandbox_defconfig-noenv-2023.07-007
cp: setting permissions for '/Tow-Boot/result': Permission denied
```

After that I have the `./result` directory created with the `dr-xr-xr-x` permissions, but it has nothing inside.

I guess, the issue is somewhere in between MacOS FS permissions, and the intermediary Linux VM, that `Docker Desktop` for Mac has to use, to run actual containers inside. And `cp` reporting error, because it can't copy files into `result` directory it just created with, because it has no Write permissions.

corrosive4354, 2024-11-04:
Since Docker Desktop uses an intermediate Linux virtual machine on macOS, it may cause file permission issues. Make sure the shared folder permissions are set correctly to allow Docker access.

Seeing that you are using Docker, I would like to recommend [Servbay](https://servbay.com) to you, which is more professional and easy to use than Docker on Mac

samueldr, 2024-11-04:
(Transparency report: deleted commercial spam from a user that is now blocked.)