Tow-Boot/Tow-Boot #345: [Request] Add Pine Rock64 support

reukiodo, 2024-11-15:
It would be awesome to get the same TowBoot used on the other Pine products also running on the old Rock64 board in SPI as well.

I see that artemis was able to get a uboot compiled and working, though didn't mention from SPI: https://artemis.sh/2022/05/16/everfrees-armferno.html

I realize it might be more involved that the Pinebook and Pro as those have static screens and keyboards, where any HDMI screen and any USB keyboard could be plugged into the Rock64, though similar in those regards to the RockPro64.

samueldr, 2024-11-16:
See also:

 - #151
 - #155
 - (and apparently #322)

Mainly blocked on the same [personal circumstances (see the preface)](https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2023.07-007) that made/makes it harder for me to dedicate time to this project.

closed 2024-11-16 by samueldr

reukiodo, 2024-11-17:
Ah very sorry! I had searched and did not come across any of those in my searching.

Thanks for all the links as it provides a LOT of reading to do. From a cursory read it seems that the main holdover is that the HDMI driver is missing? (maybe gross over simplification)