Tow-Boot/Tow-Boot #346: Current release / git version is broken on Radxa Rock 4B Plus V1.73

ryjelsum, 2024-11-22:
Hello, I've been attempting to install tow-boot onto my Radxa Rock 4b Plus. I have attempted both the shared storage strategy on eMMC and SPI installation (flashed via flashrom, using a DIP SPI flash chip) and have got a similar result both times, using both the current release version and a version built myself from git:

```
jrg@tremere ~/os-img/radxa-RockPi4-2023.07-007$ picocom /dev/ttyUSB0 -b 115200 

[...]

Tow-Boot TPL 2023.07
lpddr4_set_rate: change freq to 400MHz 0, 1
Channel 0: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=1024MB
Channel 1: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=1024MB
256B stride
lpddr4_set_rate: change freq to 800MHz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

Tow-Boot SPL 2023.07
jedec_spi_nor flash@0: unrecognized JEDEC id bytes: ff, ff, ff
jedec_spi_nor flash@0: unrecognized JEDEC id bytes: ff, ff, ff
Trying to boot from MMC1
## Checking hash(es) for config config-1 ... OK
## Checking hash(es) for Image atf-1 ... sha256+ OK
## Checking hash(es) for Image u-boot ... sha256+ OK
## Checking hash(es) for Image fdt-1 ... sha256+ OK
## Checking hash(es) for Image atf-2 ... sha256+ OK
## Checking hash(es) for Image atf-3 ... sha256+ OK
## Checking hash(es) for Image atf-4 ... sha256+ OK
spl_load_fit_image: Skip load 'atf-5': image size is 0!
NOTICE:  BL31: v2.10.0	(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980
Device tree error at node '__symbols__'
Some drivers failed to bind
initcall sequence 00000000002a3c38 failed at call 0000000000221728 (err=-11)
### ERROR ### Please RESET the board ###
```

(The addresses given by 'initcall sequence' vary by storage device but are consistent across boots)

To narrow down the issue, I downloaded the release version of Tow-Boot where support for this board was added, which is 2021.10-004. This version of Tow-Boot was able to boot my board. These are Radxa Rock 4B+ v1.73 boards. They have soldered EMMC and do not have SPI flash populated.

I will continue to bisect this in the future and figure out exactly where this issue occurs, as I have a number of these boards, and I'm using this project as a springboard to learn more about boot firmware.

ryjelsum, 2024-11-30:
Just to provide an informational update: I have since built 'mainline u-boot' 24.10 using Nixpkgs' version (`nix-build '<nixpkgs>' -A pkgsCross.aarch64-multiplatform.ubootRockPi4` provides you with a result symlink with idbloader.img and u-boot.itb, I followed [u-boot's directions](https://github.com/u-boot/u-boot/blob/af69289d61876d8e62449ee2da2dc6683bcb8198/doc/README.rockchip#L486) from here, installed to eMMC) and this issue does not occur. So, AFAICT, this is either an issue with u-boot on this board that occurred in 23.07 and was fixed sometime between then and the 24.10 release, or an issue with Tow-Boot's patches that presents itself on this particular board. 

I will try to build unpatched u-boot 23.07 using the Tow-Boot tooling once I've figured it out (I'm not great at actually comprehending Nix as a build tool) as I've seen that this is an implemented feature. I'm interested in seeing this through, I've just got other things going on and this is what I'm doing for fun :)

FooSensei, 2025-03-27:
Have you managed to fix this? I have the exact same problem with my older Rock Pi 4B 1.3. Tried building Tow-Boot from fresh Development branch too, but the result was the same. Maybe building it with current (or just more recent) u-boot codebase would fix the issue?