Tow-Boot/Tow-Boot #351: fwupd support

reukiodo, 2024-12-06:
From what I can gather, https://github.com/Tow-Boot/Tow-Boot/issues/129 is to enable future support for updating using fwupd, as it will need to be able to track versions. From https://github.com/Tow-Boot/Tow-Boot/issues/191 it seems that https://github.com/DylanVanAssche is working on getting the fwupd pieces in place, at least for the PinePhone. https://github.com/fwupd/fwupd/issues/4294

Is there a central place that tracks (maybe in a matrix) which devices are supported and which are being worked on?

I have also searched but cannot find what I should be expecting the UI to look like once a device is supported. This is what the UI shows currently on my PineBook Pro:
[image]

I assume that the 58200935-34d1-5f0f-bdb3-4a74e8f21ebb (MTD\VENDOR_pine64&PRODUCT_Pine64-Pinebook-Pro&NAME_api0.0) would somehow need to be connected to TowBoot to then show the versions available to flash?

DylanVanAssche, 2024-12-07:
Hi!

I have done some work in the past on this, but didn't continue anymore due to lack of time and other things.
There are some parts upstream, but still needs proper releases of Tow-Boot with SWID versioning data.

reukiodo, 2024-12-09:
I might be misunderstanding, but isn't this now included in the most current 2023.07-007 build?

DylanVanAssche, 2024-12-10:
We also need the firmware on LVFS of fwupd.org to make it available which requires packaging Tow-Boot as a CAB archive file with metadata.

reukiodo, 2024-12-11:
Does that mean a change/addition to the github actions build process here?

DylanVanAssche, 2024-12-12:
That requires an account on LVFS (I think it is there already), but also uploading it there manually, passing verification/tests, and have proper fwupd integration (tested with each release).